black-parrot/black-parrot #27: [PPA] bp_be_pipe_stage_reg_s and bp_be_instr_operands_s pruning

taylor-bsg, 2019-03-05:
Currently a bunch of information is carried down the pipeline that results in many additional flops. To improve energy, we will need to customize the structure (for example, pruning bp_be_instr_operands_s) as it flows down the pipeline so that it contains only information that is needed in the future. For debugging, we may want to add an auxiliary pipeline structure that contains some of this additional information, that can be optionally enabled. We may also want to consider delaying calculation of some values (for example, the immediate), so that we use fewer bits to carry it down the pipeline.

https://github.com/black-parrot/pre-alpha-release/blob/3e97c60fe6b7129871d98436f8cfb147c8695dc0/bp_be/src/v/bp_be_calculator/bp_be_calculator_top.v#L171

taylor-bsg, 2019-03-05:
The pipeline register is currently about 25% of the non-RF sequential cells in the design (1890 flip flops out of 7626 flip flops.)
  
       bsg_dff_width_p1890_0                                                                                                                                                          1872         12       1860        46315.93        33.18     46282.75

dpetrisko, 2019-03-05:
My thought was that if we left the commit tracer output port unconnected (like we should for synthesis), it would optimize out the unused bits. 

I’ll experiment and if it doesn’t, I’ll refactor into the debug pipe.

taylor-bsg, 2019-03-05:
Ah -- this might be an issue in the flow. Historically, in the designs directory, we put a top-level module that instantiates the design with the relevant parameters and the inputs and outputs wired up at necessary. But I think the setup we have now in EE477 (possibly inherited from the bp repo) is taking a shortcut and skipping the top-level instantiation, and directly synthesizing bp_be specifying parameters on the command line. 

The greater question is what we should rely upon in terms of synthesis capabilities, since we are trying to enable that eco system. Should we basically say that if Vivado, DC, and DC->yosys all successfully optimize, then we are good?

taylor-bsg, 2019-03-05:
Actually, I am thinking a little more on this, and I think the balance shifts to being more explicit about debug logic versus what is intended to be in silicon, and using a default-off parameter (e.g. debug_p=0) to explicitly enable it if we want it.  This question also comes up with simulation -- if people are incorporating BlackParrot into an SoC, then they will be running our code in VCS/verilator, they would also probably want to disable the trace mode but they won't know our core well enough to know to turn it off.  With this said, we probably want to be able to pull some information out of gate-level-simulation to know what is going on (and so our testbenches still work), so if there is information in the commit stage that can be exposed to the debug port "for free", then we probably should. I still suggest a uniform interface so we don't have to have different testbenches for the two modes, so we can just output zeros for fields that have not been carried forward when not in debug mode.

dpetrisko, 2019-03-05:
Yeah, that's reasonable.  Will update

dpetrisko, 2019-03-21:
Fixed in latest dev

closed 2019-04-04 by dpetrisko