black-parrot/black-parrot #32: [PHYSICAL_DESIGN] Logical and Physical Hierarchy Should Match

black-parrot, 2019-03-05:
In order to use hierarchical physical design in ICC, we need to have the logical and physical hierarchies align. In this case, we would like to have a single module containing a black parrot FE and BE, which corresponds to a core. Similarly we will have a single module which is a CCE. The black parrot multicore module should then instantiate N cores and M CCE's. This will allow us to do physical design for a single black parrot core, and a single CCE, and then replicate it. All personalization of the CCE and core modules according to their position or their ID should be done via external wiring rather than by parameters.

https://github.com/black-parrot/pre-alpha-release/blob/3e97c60fe6b7129871d98436f8cfb147c8695dc0/bp_top/src/v/bp_multi_top.v#L130

dpetrisko, 2019-03-05:
Why is it better to have a core with FE-BE and stamp out cores versus stamping out FEs and BEs?

black-parrot, 2019-03-05:
Adding hierarchy introduces some complexity, and makes it harder to co-design the FE and BE floorplans  (which we know will be in 1:1 ratio) and timing paths. My assumption is that the FE is not that large (or should not be that large, once we are done with it), but if it is large compared to the BE then separating them would begin to make sense. Separating out the CCE makes sense because it is on the other side of a NOC, which is explicitly designed to handle large inter-chip distances.

tommydcjung, 2019-03-06:
should NOC be separated out from bp_me_top? here is my proposed hierarchy.

[image]

taylor-bsg, 2019-03-06:
I think this is fine for now. We will see what happens with the coherence network.

closed 2019-03-07 by tommydcjung