black-parrot/black-parrot #33: proc_cfg

tommydcjung, 2019-03-06:
I'm not sure if this proc_cfg struct is really necessary. It just creates more work.
First, I have to find which file defines the struct, declare the struct. BE does not even use icache_id.
Why not just pass in cache id/core id by ports?

taylor-bsg, 2019-03-06:
Tommy, you are basically saying that instead of having a single port with the struct, to run the individual fields into the processor through separate ports? I think the idea is that there may be a lot of these kinds of parameters (for example, the start PC.)  Maybe adding a comment to the code that mentions what proc_cfg is for, and where it is defined would address the major issues?

dpetrisko, 2019-03-06:
We probably don’t need this at all once the manycore bridge is implemented

tommydcjung, 2019-03-06:
I think the intention is good, but I don't think the current use case doesn't justify proc_cfg. It's just too much work. Once, I declare the struct and I have to fill it with values and pass it to the submodules. and inside each submodule that uses proc_cfg I have to include/declare and so on... just passing as ports individually easier in my opinion

dpetrisko, 2019-03-06:
Each one of those ports will require the same declaration as the original proc_cfg, since they depend on the same parameters. Changing to ports would actually more declarations in total.

dpetrisko, 2019-03-06:
But, if all of these configuration parameters become set based on the manycore bridge bus, this is a moot point, since it will be just one address and one data port threaded throughout

taylor-bsg, 2019-03-06:
@tommydcjung I see. If we made the struct fixed size, then we would not have to do all of this struct declaration stuff, but ultimately we are passing around a bunch of fields in this struct that are not used by all of the modules which is also not desirable. Dan's proposal of using the bridge bus (which I assume is an addr + data pair) would address this, although it does mean that these items are time-multiplexed, which means for even simple usage, they may need to do a state machine (or a trace replay!) to configure it, which is not the end of the world, but a slight complexity. And it will use flops instead of wires for true constants (i.e. network IDs, and maybe hart ID) that we will worry about clock gating. (It does make you wish you had a verilog construct which was, take this struct, and tack this new stuff on to, and pass it up. Or, remove this field from the struct, and pass the new struct along.)

dpetrisko, 2019-06-18:
Keeping this for IDs.  May make sense to have a fixed width for maximum num cores/lces/cces, because then the backend flows wouldn't need to change harden memory sizes, etc. for different #core configurations.  In fact, we could even reuse bp_tile for both a 4 and 8 core hierarchical build

closed 2019-06-18 by dpetrisko