black-parrot/black-parrot #37: [CLARITY] local parameter cleanup

taylor-bsg, 2019-03-06:
Local parameters in module signature need to be cleaned up. 

vaddr_width_p and vaddr_width_lp exist in the same parameter list? -- what is the difference?  

For parameters that are not required to set the widths of inputs or outputs to modules, they should be in the body of the module.

https://github.com/black-parrot/pre-alpha-release/blob/3e97c60fe6b7129871d98436f8cfb147c8695dc0/bp_fe/src/v/bp_fe_top.v#L35

taylor-bsg, 2019-03-06:
@spikemar @zaazad

dpetrisko, 2019-03-06:
> For parameters that are not required to set the widths of inputs or outputs to modules, they should be in the body of the module.

Should I add this to the code aesthetics doc? I personally don't follow this rule and declare every constant in the header

taylor-bsg, 2019-03-06:
Good suggestion. I added it to [BSG SystemVerilog Coding Guidelines](https://docs.google.com/document/d/1xA5XUzBtz_D6aSyIBQUwFk_kSUdckrfxa2uzGjMgmCU/edit#heading=h.q2z7z8fi014v) 
under Parameter Consistency.

I believe it does improve the code readability. I didn't see too many cases where the be has localparams defined in the header that are not used.

zaazad, 2019-03-07:
Done!

zaazad, 2019-03-07:
Done.

dpetrisko, 2019-06-18:
This is an extension of the previous issue.  Basically, anything which declares structs based on the proc_params should be using the enum rather than the values directly. Fewer lines of code, less bug potential, cleaner looking...

closed 2019-08-14 by dpetrisko