black-parrot/black-parrot #45: [CLARITY] Code replication on cam

black-parrot, 2019-03-06:
We have a cam in bsg_ip_cores (BaseJump STL) that seems very similar. Is there anything unique about this one? Otherwise we should use the BaseJump STL one. Or if they are different, what is the change and how should we improve BaseJump STL to accommodate it?

https://github.com/black-parrot/pre-alpha-release/blob/3e97c60fe6b7129871d98436f8cfb147c8695dc0/bp_be/src/v/bp_be_mmu/bp_be_tlb/bp_tlb_cam.v#L2

farzamgl, 2019-03-06:
"bp_tlb_cam" also outputs the first empty entry which makes the TLB write easier when there are still some empty locations.

taylor-bsg, 2019-03-06:
Okay great. Can you reconcile them and commit to bsg_ip_cores? It looks like bsg_cam_1r1w allows you to have multiple entries in the CAM, and has a few improved coding practices, but the finding empty entry could also be useful. Maybe have a parameter find_empty_p that says whether it should find the empty entries, and it can assert empty_v_o and skip the priority encoder if it is set.

farzamgl, 2019-03-06:
Should I modify the current module or create a new one? Because it might break the design of the people that are using the current version in bsg_ip_cores.

taylor-bsg, 2019-03-06:
Thanks Farzam -- not to worry, nobody is using that module yet. Zhara from BU wrote it, probably to use in the FE TLB. =)

closed 2019-03-18 by farzamgl