black-parrot/black-parrot #47: [CLARITY]: Pseudo LRU module --> BaseJump STL?

black-parrot, 2019-03-06:
This seems like a neat module. Perhaps we should have it in BaseJump STL? Does it work, or does it have outstanding issues? Also, should it be folded into the data cache as well?

https://github.com/black-parrot/pre-alpha-release/blob/master/bp_be/src/v/bp_be_mmu/bp_be_dcache/bp_be_dcache_lru_decode.v

https://github.com/black-parrot/pre-alpha-release/blob/3e97c60fe6b7129871d98436f8cfb147c8695dc0/bp_be/src/v/bp_be_mmu/bp_be_tlb/bp_plru.v#L2

tommydcjung, 2019-03-06:
there is encode and decode part. By looking at its ports, I'm not sure if it does the same thing as in dcache_lru modules. Also this should be a combinational logic.

farzamgl, 2019-03-06:
This is basically the encode, decode and the LRU register in a single module. I suppose we can separate them if needed and try to use it in the d-cache as well.

taylor-bsg, 2019-03-06:
Farzam's code handles the general case of n-way though, or is that not true?

farzamgl, 2019-03-06:
Yes, it's parameterized for N way.

taylor-bsg, 2019-03-06:
Cool. Maybe refine the interface with Tommy and see if you can create something that would generalize well for BaseJump STL and work in these two use cases.

closed 2019-03-15 by tommydcjung