black-parrot/black-parrot #48: [PPA_Medium] Missing clock gating?

black-parrot, 2019-03-06:
Looks like this SRAM is always on.

https://github.com/black-parrot/pre-alpha-release/blob/3e97c60fe6b7129871d98436f8cfb147c8695dc0/bp_fe/src/v/bp_fe_btb.v#L63

zaazad, 2019-03-07:
Once ICache data (instruction) is ready, based on the instruction's type we decide whether to use btb's output to calculate the next_pc or not. If so, then read enable signal should be activated. However, it needs to be set one cycle ahead since memory is sync.
So I think this signal should be set whenever there is either a write operation or a valid cache read request in the previous cycle (it takes two cycles to get ICache data for the provided PC).

dpetrisko, 2019-04-04:
No longer present in dev

closed 2019-04-04 by dpetrisko