black-parrot/black-parrot #53: [PHYSICAL_DESIGN] inter-router links should be READY-AND, not READY-THEN?

black-parrot, 2019-03-06:
I noticed in the code that it signifies them as ready then and not as ready then? Is this the case (if it is, presumably it is imposed by the CCE<->network or LCE<->connections? This would reduce the inter-router distance that we can support to 1/2, which is worrisome.

https://github.com/black-parrot/pre-alpha-release/blob/3e97c60fe6b7129871d98436f8cfb147c8695dc0/bp_me/src/v/network/bp_coherence_network_channel.v#L73

closed 2019-04-22 by muwyse

black-parrot, 2019-04-23:
Hi Mark,  how was this resolved?

muwyse, 2019-04-23:
The original network design was intended (by me) to use r->v since I made some arbitrary decisions about the CCE design that made it universally demanding. However, this changed to be r&v as suggested to improve route-able distance, etc. The change happened over time (D$ and I$ revisions, CCE revisions, network revisions that combined data networks and made them wormhole routed, etc.). The specific code changes that resolve this issue are changing the macros used to properly reflect the actual protocol all the components now implement at the network module interface, which is r&v. I believe all the components properly use r&v currently, and our regression tests use the network between all the LCEs and CCEs. The fact that the network and everything connecting to it is input buffered may mean that a particular component is not r&v without the buffer, so if we work to optimize away the buffers then we will need to keep an eye out for that.

taylor-bsg, 2019-04-23:
Yep Mark, yep we can never totally remove the buffers but can downsize from
two to one elements assuming it is okay to not send a word every cycle.

M