black-parrot/black-parrot #55: GP parameter casting

spikemar, 2019-03-06:
https://github.com/black-parrot/pre-alpha-release/blob/71a28a51c186fc3d997c91f37139747cc48f2ead/bp_be/src/v/bp_be_mmu/bp_be_mmu_top.v#L91

When we use a global parameter should we cast them to local params? We pass these parameters to lower level modules as is. Should we do it this way or call the global parameters directly from those modules?

spikemar, 2019-03-06:
Also related to this, should we cast the gp to a lp instead of using it directly?

dpetrisko, 2019-03-06:
I think you can just use it directly.  I tend to use that style because it's more obvious what global packages you rely on, you can have shorter names and if you need to change the _lp to be calculated or non-global, there's only one place to change it.  But there's no technical reason not to use the global directly

closed 2019-04-04 by dpetrisko