black-parrot/black-parrot #63: [PHYSICAL_DESIGN] Coherence network needs to be parameterized.

taylor-bsg, 2019-03-07:
Current experience with the coherence network indicates that there is a lot of congestion and wasted area as a result of the wide crossbars. Possibly this could be fixed with a bunch of physical design work, but this generally makes the core non-portable. I think we should parameterize the data transfer buses so that there is a serialization factor for the data buses. So for example currently the buses are 512 bits so we would like to support a serialization factor of 1,2,4,8 and 16, so that they can be 512/256/128/64/32 bits (for example, using this module: https://bitbucket.org/taylor-bsg/bsg_ip_cores/src/master/bsg_dataflow/bsg_parallel_in_serial_out.v). This may require verifying that the protocol is still correct if the data words do not come as an atomic unit. I believe that the serialization itself should probably be placed on the LCE side, because then the LCE can (later) be optimized for the particular serialization factor. 

https://github.com/black-parrot/pre-alpha-release/blob/8a9000eb112538204d5bfb27bff71501b61558f1/bp_me/src/v/network/bp_coherence_network.v#L22

tommydcjung, 2019-03-07:
On these data buses, there are other metadata that is bundled together with 512-bit data. For example, bp_cce_lce_data_cmd_s has dst_id, src_id, msg_type, way_id, and addr (56-bit). Those infos are passed back and forth so that CCE is kept stateless. If we were to break down 512-bit data into 64-bit chunks, should those other infos be included in each of those chunks?

tommydcjung, 2019-03-07:
Also, since LCE can have only one outstanding miss, I think we could drop some of those fields in the message. For example, LCE already knows which addr caused the miss and which way to replace, so addr and way doesn't necessarily have to be part of message.

dpetrisko, 2019-03-07:
Wouldn’t you need serialization on both sides of the links?

taylor-bsg, 2019-03-07:
A) In terms of unnecessary data being transmitted: I think we need to walk through the individual message types and see what is needed on a case-by-case basis. We would prefer that the CCE not have to maintain state for each client, but it is okay to expect a client to have to remember its own data.  For data being returned to the LCE, I think it would be fine omit the address but to use a transaction ID (which is configurable in size, but we can default to say 5 bits), which should we have an LCE that can sustain multiple misses, it could use this to track things, but in the default case, the LCE uses a transaction ID of 0 and has a single register to hold this information. The transaction ID would be sent with LCE requests so that the CCE always knows which transaction ID to use.  

B) Since we are not packetizing this into a fixed size channel, it would be okay to just say that the packet is divided into N pieces, but that the dst_id is on every piece, as is the trans_id.

C) It also makes sense for us to start thinking about merging networks. A first step would be to see if 
 we can merge the two --> LCE data networks? I think this is an easy case and we can easily prove it would not cause deadlocks. In a 4 CCE, 4 LCE network, one network would 14 unidirectional links, and the other would have 14 unidirectional links, and the combined network would have 18 links.

black-parrot, 2019-03-08:
Dan — since the directory does not have the data, it is sort of up to what the L2 implements. It may not need to do a full deserialization.

dpetrisko, 2019-08-01:
Networks changed to wormhole some time ago, comments still relevant but issue is resolved

closed 2019-08-01 by dpetrisko