black-parrot/black-parrot #66: Get rid of bsg_ip_cores submodule

tommydcjung, 2019-03-07:
I understand how submodule can be useful, but bsg_manycore does not use submodule. we cannot have two different version of bsg_ip_cores when fusing bsg_manycore and bsg_ip_cores together.

tommydcjung, 2019-03-07:
@taylor-bsg

dpetrisko, 2019-03-07:
If you update the submodule to whatever bsg_manycore is using, then there is no problem.  Having bsg_ip_cores just on master has caused broken builds for us twice in the last two months. A stable version is necessary

tommydcjung, 2019-03-07:
if a build breaks, then it just has to be fixed. It's just hiding bunch of problems, and later someone finds out all these problem when trying to create bsg_designs. submodule is not solving the problem. what really should have happen is that someone who modified bsg_ip_cores and caused black-parrot build to break, should have fixed it in black-parrot.

closed 2019-03-08 by tommydcjung

stdavids, 2019-03-08:
We should discuss submodules in a group meeting at some point. They do pose challenges to the larger cad infrastructure, particularly bsg_ip_cores. Maintaining versioning of bsg_ip_cores across multiple projects is tough, but there are a couple solutions we should talk about.

black-parrot, 2019-03-09:
Seems reasonable, Scott!