black-parrot/black-parrot #84: Verification of parameterized structs

dpetrisko, 2019-03-10:
Our approach to parameterized struct declarations are a good way to allow variable size structs and ports.  However, having separate definitions for `width and `declare_s is error prone and could lead to width mismatch bugs.  We would like to have a module for each package which contains parameterized structs.   This module would be instantiated at the top level and it's entire purpose would be to `assert $bits(my_super_cool_struct_s) = `my_super_cool_struct_width` for all structs in a package.

dpetrisko, 2019-05-26:
Another idea would be to generate the struct widths from a Python script run over the struct definitions

taylor-bsg, 2019-05-26:
#Solved_by_Shylock

A related idea solving a slightly different problem is to fold these width
definitions into the config infrastructure -- we just store the widths in
that package. This would probably make the code much shorter and more
readable.

dpetrisko, 2019-05-26:
Ooh, I like that. At that point, we’d probably want to generate the configs, though, because they’d get huge

dpetrisko, 2019-07-13:
Partially solved by printing bits(struct) and struct_width_lp for bp_tile.  We can expand this solution to other modules, or do something fancier to parse them all at once

closed 2019-07-13 by dpetrisko