black-parrot/black-parrot #107: [CLARITY] Should not be forced to source setup_env.sh to run Makefile infrastructure

stdavids, 2019-03-16:
In general, we should avoid using sh scripts in favor of makefile to be consistent with the rest of our infrastructure code. But we should especially avoid requiring that a sh script be sourced for the makefiles to work. Common environment variables should be put in a common location for makefiles to include.

I personally find it cumbersome to have to source a script every time I login or open a new shell... but beyond that, sourcing is also shell specific unlike makefiles (ie. setup_env.sh only works with bash, it won't work with other common shells like csh). The other thing is that sourcing has this really bad persistence issue when developing infrastructure. When writing infrastructure code you basically want to make as few assumptions about the environmental state of the machine as you can. But when sourcing is part of your setup it become real tedious to get back to that near-0 assumption state (you need to close your shell, open a new one and re-source). If you don't then you can get into really silly issues... for example: if you want to change the name of a variable in setup_env.sh, if you don't shutdown your shell then when you re-source the script both the old and new variables will exist in your environment which can hide dependencies on the old name that were not updated.

dpetrisko, 2019-03-16:
Sounds good.  I originally did it because I couldn't figure out how to get flists to respect environment variables (but I guess you can just .export from Makefiles).  I'll refactor into a Makefile.common that sets the environment variables

dpetrisko, 2019-03-16:
(Probably should do this for the regression scripts, too, although I'm kind of waiting to see how CI will interact with them)

dpetrisko, 2019-03-21:
Fixed in 82dda501d5dd12a8d3d31cd1bc7302058693e950

closed 2019-03-21 by dpetrisko