black-parrot/black-parrot #114: Branch prediction improvement

dpetrisko, 2019-04-04:
Re-add BHT and use it to override BTB predictions e.g. predict not taken when target is in BTB

Write to BTB on branch taken only to save space. Currently, we write on not-taken branches as well.

BTB is only written on redirects at the moment. There are two ways to handle this: we could also add calculated branch targets on predicted taken branches which are not in the BTB. This may get complicated because of the 1RW BTB memory.  Hopefully, there is a solution which does not require dropping conflicting BTB writes.

black-parrot, 2019-04-04:
Todo pending the acquisition of a suite of benchmarks that provides sufficient data that we can make justifiable PPA decisions.

dpetrisko, 2019-09-26:
DSE still needs to be done for optimization, but the flexibility is there to support both BHT and BTB

closed 2019-09-26 by dpetrisko