black-parrot/black-parrot #119: Move CCE-MEM interactions to ME defines

dpetrisko, 2019-04-06:
Currently, we have cce_mem commands in bp_common_me_if, but this should only hold the LCE-CCE interface.  The CCE-MEM interface is much more likely to change or be platform specific,  whereas the LCE-CCE interface should theoretically be very stable.

taylor-bsg, 2019-04-06:
I think that the immediate CCE<->MEM cache interface should be pretty standardized; mostly looking like the wormhole-routed packet format that travels on the CCE <-> LCE networks. It will (at some point) be converted into AXI, our open source DRAM protocol, a custom L2 protocol, or the manycore network protocol.

dpetrisko, 2019-04-06:
That may be, but all of those are still entirely ME-side interfaces

taylor-bsg, 2019-04-06:
Yes, but I think what I am saying is that there is an interface that can be defined between the CCE and ME that does not change.

closed 2019-08-07 by dpetrisko