black-parrot/black-parrot #121: Standardize file header template

dpetrisko, 2019-04-06:
We should have a standard file header template with license info, name, description, keywords and notes.

Debatable whether we should have inputs/outputs/parameter descriptions in the header. There's a greater chance that they fall out of sync with the actual ports. However, it would be easier to extract from the header if we do some kind of auto-documentation in the future.

Possible candidate:
https://github.com/black-parrot/pre-alpha-release/blob/master/bp_be/src/v/bp_be_calculator/bp_be_bypass.v

somewhat loosely based on page 20 of https://people.ece.cornell.edu/land/courses/ece5760/Verilog/FreescaleVerilog.pdf

taylor-bsg, 2019-04-06:
I think the header format should focus on information required to understand the module that is not found elsewhere in the code. So for example, the description is helpful, as well as a pointer to a google design doc if applicable, or if not, a description of the "philosophy of operation" of the module, which might include invariants, preconditions, and postconditions.  I think I would prefer not to replicate the parameter and port list since it will get out of date, but rather to put the relevant comments inline with the actual ports.

dpetrisko, 2023-07-18:
Closing as not useful

closed 2023-07-18 by dpetrisko