black-parrot/black-parrot #123: reorg module hierarchy so bp_core includes network routers.

tommydcjung, 2019-04-10:


dpetrisko, 2019-04-10:
Which routers is this referring to?

tommydcjung, 2019-04-10:
talking about the routers inside the me networks.

dpetrisko, 2019-04-10:
What’s the motivation to move them to core?

tommydcjung, 2019-04-10:
we want to have BE+FE+CCE+routers as one hierarchy node in physical design, so we can stamp them out in tiled fashion. Just like we include router in manycore tile.

dpetrisko, 2019-04-10:
But that’s bp_top. What is in ME in this organization?

taylor-bsg, 2019-04-10:
bp_top can instantiate bp_tile; the network links will be the main
interface for bp_tile.

dpetrisko, 2019-04-10:
Wait, what’s bp_tile now?

taylor-bsg, 2019-04-10:
The replicated unit of the multicore; contains 1 CCE, 2 routers, 1 FE, 1 BE.

dpetrisko, 2019-04-10:
So bp_core becomes bp_tile? Or bp_tile instantiates bp_core and bp_me_top?

taylor-bsg, 2019-04-10:
bp_tile could instantiate two routers, one bp_core, and one CCE.

closed 2019-05-08 by dpetrisko