black-parrot/black-parrot #124: Enhancements to BP Mem

dpetrisko, 2019-04-10:
We want to change BP mem to 1) read in from plusargs instead of boot rom and 2) support more than 15 bit addresses. When we run larger benchmarks, this could become an aliasing issue. 

Outside of bp_mem, but related, we want an option to load the trace rom dynamically so we can avoid recompiling for each test.

dpetrisko, 2019-04-10:
@muwyse

taylor-bsg, 2019-04-10:
Medium term, we may want to integrate dramsim2 or the like to simulate
memory timings, and an L2.

As far magic goes, it is better to have it
done in the top level testbench rather then in a leaf model so it is easy
to track down.

M

dpetrisko, 2019-04-10:
Bp_mem is in the testbench, are you saying we should flatten it and put the functionality directly in the Tb?

taylor-bsg, 2019-04-10:
Or pass the file name to load into the module

dpetrisko, 2019-04-10:
Oh, I see what you mean. Yeah, that’s the plan.

dpetrisko, 2019-04-10:
Actually, that wouldn’t work — if we pass the filename as parameter then we would still have to compile between runs

taylor-bsg, 2019-04-10:
Not as a parameter as an input

dpetrisko, 2019-04-10:
Interesting idea. Is there any bsg IP that does it this way?

taylor-bsg, 2019-04-10:
Here is some code to look at. At some level, we might even want to reuse/ (slightly refactor) this module, since at first cut, the memory system for both BlackParrot and HB manycore is the same, they both use RISC-V binaries, both have DRAM and victim caches attached to the manycore network.

Adding @tommydcjung since he is doing the width that connects BlackParrot to the memory system.

https://bitbucket.org/taylor-bsg/bsg_manycore/src/master/testbenches/common/v/bsg_manycore_spmd_loader.v

Getting back to parameter versus input, this actually uses the parameter, but it just means when you run, you have to run in a different directory from your VCS binary; you don't have to rebuild. I think a long time ago when I wrote the initial version of this module, I tried to pass a filename in as an actual module input and VCS did unexpectedly bad things. Could be fixed.

dpetrisko, 2019-04-10:
Cool! I think parameter is fine if we copy in the right binary.

How about the trace-replay rom?  How do you feel about instantiating an unpacked array, readmemh, and connecting it to the trace replay node?

taylor-bsg, 2019-04-10:
It’s okay, but why?

Generally we will have 1 benchmark per directory, so no need to copy in a
binary per se; we will be in the dir where the binary is generated.

M

dpetrisko, 2019-04-10:
Agreed.  right now, the binaries are pre-generated so we have to just pick one and copy. 

Because otherwise we have to recompile for each TR test -- each ISA test + benchmark.  I guess the better solution would be to have the TR ROM have the entire regression run in a single trace.

taylor-bsg, 2019-04-11:
With the right make rules we should not have to actually recompile in
successive runs. For small tests whatever is fine, but for real benchmarks,
it is too cumbersome to pregenerate, and pretty small. Is it easier from a
regression standpoint for things to be separate for debugging and for
trying out different variants of execution.

dpetrisko, 2019-04-11:
I agree the binary generation should be separate.  It's on the todo list.

I don't see how make rules can save us from recompiling when we change out trace roms. It's a new verilog module in the testbench, so VCS needs to recompile it into the simv.  Unless there's a way to dynamically link it

taylor-bsg, 2019-04-11:
We should not use trace rom for executing
programs...

dpetrisko, 2019-04-11:
We don't.  We use them for checking results of running programs

dpetrisko, 2019-04-11:
This is how the regression works -- trace roms generated from running spike

taylor-bsg, 2019-04-11:
Programs should be self-checking and output PASS, DONE and FAIL codes. PASS
says a sub test worked, DONE says all sub tests worked and FAIL says a sub
test failed and no more subtests were run, execution terminates. The output
is written to a log file in a log directory so a makefile rule can harvest
results.

What is the part about spike checking? Can you output a file and then
compare the files?

dpetrisko, 2019-04-11:
Having programs be self checking is a fairly big burden IMO. For instance, for the FE, completion is when PC=last pc, but success is that each of the correct PC gets fetched. As another example, I don’t believe the riscv benchmarks are sufficiently self-checking. Lastly, that won’t catch subtle bugs. 

Right now spike is run and transformed into a trace rom. Then the program is run and each commit from BP is checked against the next trace rom entry. This way, each commit is verified.

We certainly could print out each commit and compare to spike after the test. I think that approach is better (with the exception of it making it a little more difficult to track down mismatches, since it will allow wrong path execution to continue forever). However, currently, we only have a testing set up to do the dynamic checking. 

So there are really 3 action items here:
1) have bp_mem readmemh and cover a larger range of memory
2) enable flow to log commit output and compare to spike output after test run
3) create binaries and spike traces on demand

taylor-bsg, 2019-04-11:
Yeah the testing infrastructure verification varies a little bit.

The spike stuff is pretty sweet if we can match it. Moving tests to
comparing output trace files is useful since it allows offline analysis,
and we can use a diff tool to compare them (Unix diff is nice but does not
scale to large files but I have a layer on top of it that I can dig up from
the raw days).

For the self checking stuff, it is mostly for either unit tests that we
write or for checking high level output properties (is the array sorted?)

The spike comparison stuff is nice because spike is pre-verified so we have
a golden model; for Raw we were testing the golden model and our RTL, both
which could have bugs. :-)

I guess the place where unit testing requires more manual self checking is
where we chose a different implementation versus spike (floating point?
Exceptions? Emulation PMP? Etc). Here, a smart diff implementation can help
resynchronize the trace after divergence.

M

dpetrisko, 2019-04-11:
> I guess the place where unit testing requires more manual self checking is
where we chose a different implementation versus spike (floating point?
Exceptions? Emulation PMP? Etc). Here, a smart diff implementation can help
resynchronize the trace after divergence.

++ to this. We've already had to do hacky things to get around differences in PMPs, etc.  Having a more solid, well thought-out approach would be very valuable.  Perhaps we can investigate SWERV's approach, since they have some flavor of co-simulation (although I believe they rolled their own simulator rather than spike) https://github.com/westerndigitalcorporation/swerv_eh1

muwyse, 2019-04-11:
Back to the very first note: are you saying we need changes to bp_mem.v? This module can be completely reworked as needed; it was never meant to be a real memory in any way, rather a simple, small(ish) thing that talks the CCE-MEM interface so I could debug the CCE itself. Or perhaps, it may be useful to create a better module for use with regressions, and I can keep this module as is to act as a fake memory for my CCE only testing.

dpetrisko, 2019-04-11:
Yeah, the most pressing thing is to have bp_mem readmemh from a filename rather than a boot rom.  

That change itself is not that bad, but there’s tons of infrastructure changes that need to happen to make it efficient, including changes the trace replay, spike diffing and Makefile flow

dpetrisko, 2019-04-21:
Closing this issue, because it morphed into #134 and #135

closed 2019-04-21 by dpetrisko