black-parrot/black-parrot #128: Committed binary files

dpetrisko, 2019-04-15:
There were some committed and deleted binary files in branch be_dev_vm_tests.  We don't want these to end up in our git history, so  we need to rewrite the history of this branch.  I'll close this issue once those binaries are gone from the history.

iihihiuh, 2019-04-16:
Just updated. Please, see if this work.

dpetrisko, 2019-04-16:
Hm, it seems a little strange, there are like 250,000 changed lines.  It kind of looks like this commit is totally dissociated with the previous history

iihihiuh, 2019-04-16:
I created another branch "be_dev_vm_test" that spin off from be_dev. History record seems good to me. Branch "be_dev_vm_tests" is removed.

dpetrisko, 2019-04-16:
Okay cool, history looks good. What is test.S.main.S?  It's really big, is it the output of running some script? If so, we should just commit the script

iihihiuh, 2019-04-16:
Are we allowed to add an empty folder in the repository?

dpetrisko, 2019-04-16:
No, git does not allow you to do so.  Why do you need one?

iihihiuh, 2019-04-16:
Actually, I do not need it. Thanks. Everything should be fine by now.

closed 2019-04-17 by dpetrisko