black-parrot/black-parrot #131: limit the number of outstanding uncached store requests.

tommydcjung, 2019-04-19:


taylor-bsg, 2019-04-19:
Also related to this; since this is likely to be implemented with a credit counter:

Implement fence instruction that allows us to confirm that we have no outstanding credits.

M

tommydcjung, 2019-04-24:
The problem is that once uncached store is sent out, there is no ack coming back to LCE to restore the credit.

dpetrisko, 2019-04-24:
If there’s some guaranteed latency by which you’re guaranteed to have completed, you could increment based on that?

tommydcjung, 2019-04-24:
maybe. uncached store will raise miss, if the network was busy, and it will have to be replayed and retry. what if we increase the delay between each retry by some increasing function?

tommydcjung, 2019-04-24:
but that still doesn't solve the problem with not knowing when to restore the credit..

dpetrisko, 2019-04-24:
I'm saying you can restore the credit when a cycle counter exceeds the maximum latency

black-parrot, 2019-04-24:
All stores should have an acknowledgement packet, and there should be a counter that counts them.  There is no deterministic number of cycles after which we know a store has completed in the I/O system. Since loads have a response packet, there must be a pathway to implement a store ack packet.

dpetrisko, 2019-04-24:
That works, too

taylor-bsg, 2019-04-24:
It will give you a reason to implement fence, .rl and .aq. ;-)

dpetrisko, 2019-05-25:
Added in be_dev 548b0b6

closed 2019-05-25 by dpetrisko