black-parrot/black-parrot #133: Fork and submodule RISC-V tests

dpetrisko, 2019-04-20:
Our version of RISC-V tests is very out of date and does not have newer tests such as CSR and PMP tests, which are valuable. This change needs to be compatible with our existing testing infrastructure I.e. linker scripts and start PCs

taylor-bsg, 2019-04-20:
There is also this -- not sure what the relationship of this repo is to
others:

https://github.com/riscv/riscv-compliance/tree/master/riscv-test-suite

M

dpetrisko, 2019-04-20:
Oh interesting. Looks like a superset of riscv-tests. 

I’ll also add this to the issue:
https://github.com/google/riscv-dv

Someone should investigate how stable / usable it is for us. 

It’s SV-UVM so it’s not verilator compliant, but it should be fine with VCS.

dpetrisko, 2019-05-25:
riscv-dv seems great. It has ways to generate subsets of instructions you support, random MMU stress testing, etc. The only question mark I have from playing around with it is how to link in our emulation code (i.e. atomics), since they set up the trap vector for us.

dpetrisko, 2019-05-25:
Probably the thing to do is to set up the environment to test RV64I instructions (for which we have full support) and at some point figure out how to inject our emulation code into their instruction generator to test atomics (and supervisor mode (and mul/div/fp)))

dpetrisko, 2020-04-26:
Closing this, as they are submoduled and we're tracking riscv-dv in another issue

closed 2020-04-26 by dpetrisko