black-parrot/black-parrot #138: Create issue and PR templates

dpetrisko, 2019-04-20:
Currently Issues and PRs are loosely organized.  This works well for internal issues and PRs because we can just talk to the owner for clarifications.  To effectively solicit external contributions (Virality), we should guide people towards making complete issues and PRs.

https://help.github.com/en/articles/about-issue-and-pull-request-templates

For example, Issues should have 1) Scope of change 2) Affected users 3) Steps to completion 4) Potential blockers, 5) etc....

PRs should have 1) Scope of changes 2) PPA analysis 3) Testing strategy 4) Requested feedback 5) etc....

taylor-bsg, 2019-04-21:
I think the high level goal is good, but I think the key question is
“when”. I am reminded of the Y Combinator aphorism “build things that don’t
scale” or Facebook’s “move fast and break things”.  One aside; we are using
issues as a bug reporting tool, so in that case it should suffice just to
give the bug?

Adding more process does slow down contributions and also de-energizes
contributors by increasing the activation energy. I think the key question
that it comes down to is, whose time are we saving with additional process
steps? Since we have no users, it is not their time. :-).

A near term approach is to say:

- if you break the build, we expect you to drop everything and fix it
immediately.

- PPA optimizations should be avoided unless they significantly move the
needle on PPA and do not increase the code size very much at all.
Functionality is our focus.

- the regression should always be run before committing code

- if you are adding a new component, you must add the code that tests it to
the regression.

- developers control their chaos by periodically updating to the latest
when they are prepared to do some debugging work that may be required at
that time.

M

dpetrisko, 2019-04-21:
I don't disagree with what you're saying, but I think it's orthogonal to this issue.  The template is a time saver that prompts users (or us) from asking the same questions repeatedly.  What you're saying is we should keep the template lightweight (and not require PPA reports, yet)

taylor-bsg, 2019-04-21:
Yep and not be persnickety if they omit something that, in that particular
situation is not that important.

M

dpetrisko, 2019-04-21:
Oh absolutely.  Be Friendly :)

dpetrisko, 2019-04-21:
> One aside; we are using
> issues as a bug reporting tool, so in that case it should suffice just to
> give the bug?

Also, having a few issue templates can help keep them trim: i.e. 1 for bugs, 1 for features.  Bug reports should include reproducibility steps (what test fails or if it's just an identified bug, what conditions will trigger it).

taylor-bsg, 2019-04-21:
Like