black-parrot/black-parrot #140: Atomic support for uncached regions

dpetrisko, 2019-04-20:
We need to consider atomics for uncached regions, namely the manycore system. The RISC-V method for determining which regions have support which types of memory access is through PMAs. We should store PMA properties (uncached, atomic supported, pre-fetchable) in the TLB.  We also need a way for software to query PMAs, most likely through MMIO.

As for the actual atomic access, my understanding is that the manycore network supports this functionality, so we need a method of passing memory request info through the manycore bridge.  Since these are uncached accesses, I don't believe BlackParrot needs any additional hardware support, but we need it to speak the same interface language.

At the moment, we throw an assertion if an atomic operation is performed on an uncached region.

Action items are 
1) Design and implement an enhancement to the manycore bridge to support atomics
2) Add atomic interface information to manycore bridge documentation.
3) Add regression test for this interaction

See additional discussion in PR #129

taylor-bsg, 2019-04-21:
Mostly agree except the part about PMA being stored in the TLB, since PMA
must apply even when translation is off. I think we should see how far just
using address ranges gets us.

dpetrisko, 2019-04-21:
I thought we agreed that we were filling the PMA along with the PMP, even with translation off.  Otherwise the FE will have no conception of the PMAs

taylor-bsg, 2019-04-21:
We explored that idea and then agreed to punt until later. I think we can
have a module that is reused in both the FE and the BE that returns the
appropriate information, which is currently statically determined?

dpetrisko, 2019-04-25:
Getting back to this:

Do you think we'll need AMO support for IO, or just the LR/SC? Currently how the emulation works is we detect AMO as illegal instruction at decode.  It wouldn't be too much trouble to add that detection to be after physical address translation -- AMO+cached is illegal, AMO+uncached is legal (pass it to the CCE and beyond).

taylor-bsg, 2019-04-25:
Interesting. Yeah, we would like AMO support for I/O; for the manycore in
particular.

M

dpetrisko, 2019-04-26:
In the riscv spec PMA section, they describe AMO support in terms of subsets of supported operations. The least of which is supporting only AMO.swap. What would be nice about that is we wouldn't need an ALU in the endpoint controller. Do you think that’s sufficient?

taylor-bsg, 2019-04-26:
That’s interesting. We haven’t totally settled on what we support, but
ideally it will be bottom up.

Does the latest spec have a lot more on PMA?

M

black-parrot, 2019-04-26:
[image]

I'm not sure when this was introduced, but it's in latest

dpetrisko, 2019-06-18:
So it turns out it's actually easier to have PMA attributes live in/near the PTW.  Otherwise, we need to manually map an MMIO page as part of our start code #203 .  And I'm not sure how this would interact with Linux (this is basically the IOMMU concept).  The only real downside is that the FE-BE interface needs to contain the PMA information (tlb_fill_response), but there aren't that many PMA bits so it should be scalable.