black-parrot/black-parrot #144: Processor Visualization + Debugger

dpetrisko, 2019-04-24:
It would be nice to have a way to visualize our pipeline, along with events.  Currently, we have a few tracers printing logs (Calculator and CCE), but having a way to look cycle by cycle at what's happening all over the processor would be nice.

There are a few ways to handle this.
1) have a timestamp on each event in the log files, then postprocess them.
2) have log events go to DPI

DPI is appealing because we can reorder, post-process, choose to pipe to file or std::cout, with the power of C/C++. This (or similar) is how I've seen industrial-strength debuggers implemented.

TLDR; VCD dumps are awful for debugging. Let's figure something out before we try to boot Linux.

dpetrisko, 2019-04-24:
Additional consideration: how do we log emulated instructions?

taylor-bsg, 2019-04-25:
What do current traces that we use to compare to spike contain?

We have a similar need to have some interactive visibility into the
manycore. Makes sense to solve it in a similar way.

Regarding DPI — what about VPI? The way I see it is we can map in values
via the module hierarchy and then create GUI panels (which includes
curses!) that display those values?

Then there is a facility to specify some stop conditions and/or have some
basic interactive commands like single step, go, open this panel, etc.

I think this is useful for tracking down anamolous microarchitectural ticks
but maybe not for long running programs like Linux.

M

dpetrisko, 2019-04-25:
> What do current traces that we use to compare to spike contain?

The comparison is based on register writes and memory writes.  Additionally, we dump pc, instruction, operand values, and the operation.  I also print out some events such as poisoning, rollbacks, traps.

> Regarding DPI — what about VPI? The way I see it is we can map in values
via the module hierarchy and then create GUI panels (which includes
curses!) that display those values?

Yeah, you're right.  I incorrectly thought DPI was a superset of VPI, but looking into it people use VPI directly for panels. (Do you know of any academic processors which have panel support? That could be a neat feature proposition)

> I think this is useful for tracking down anamolous microarchitectural ticks
but maybe not for long running programs like Linux.

Definitely.  I more meant debugging complicated (probably random) programs to root out bugs before trying to boot Linux. The ideal flow is something like: run on FPGA (or ASIC), find the bug, dump state, replay in simulator with panel support.

taylor-bsg, 2019-04-25:
This document talks about DPI versus VPI:

https://s3.amazonaws.com/verificationacademy-news/DVCon2016/Posters/dvcon-2016_introspection-into-systemverilog-without-turning-it-inside-out_poster_paper.pdf

Seems like DPI is supposed to be the thing after VPI.

But part of it is, for an interactive thing, it would be nice to pull the
values out of the simulation if they are being viewed, rather than to have
the simulator push them out, even if they are not being looked at.

Here is some sample VPI code that reads the Verilog variable "
testbench.u1.u2.dout"

# include "vpi_user.h"

vpiHandle net;              // name of signal to probe
s_vpi_value cur_val;  // current value of the signal

net = vpi_handle_by_name("testbench.u1.u2.dout", NULL);  // set the name of
the signal you are probing
vector_size = vpi_get(vpiSize, net);    // returns the size of the signal
you are probing
cur_val.format = vpiVectorVal;             // set the format you want for
the returned data
vpi_get_value(net, &cur_val);              // probe the signal and return
the value into cur_val structure

// The value of the probed net is available in:

cur_val.value.vector[0].aval   // bits [31:0] of the probed signal
cur_val.value.vector[0].bval
cur_val.value.vector[1].aval   // bits [63:32] of the probed signal
cur_val.value.vector[1].bval
... and so on, depending on the vector_size of the net you probed

M

taylor-bsg, 2019-04-25:
Dan -- sounds like we have converged.

Regarding panels; I didn't know that was a term for these things. But we
when we were coding VirtualPC (dynamic x86 to PowerPC translator), we found
it was a really nice abstraction to be able to pull up dialog boxes that
displayed the state of various pieces of HW (e.g. sound card, video card,
interrupt controller), and to view and even edit it.  I think ultimately I
wrote some kind of parser that would allow you take in a struct and
automatically lay out a panel for it to handle the common case.

It would be cool to have a bsg package that allows you to panelize your
HW...

We had a lot of discussion with Prof. Batten about what we actually get
from an ISA simulator versus just using RTL. The first was the GUI and the
second was just being able to localize a bug. I think the panels
abstraction addresses the second item and could make "Direct-to-RTL" much
more palatable to software types.

M

dpetrisko, 2019-04-25:
> Regarding panels; I didn't know that was a term for these things. But we
when we were coding VirtualPC (dynamic x86 to PowerPC translator), we found
it was a really nice abstraction to be able to pull up dialog boxes that
displayed the state of various pieces of HW (e.g. sound card, video card,
interrupt controller), and to view and even edit it.  I think ultimately I
wrote some kind of parser that would allow you take in a struct and
automatically lay out a panel for it to handle the common case.

It's a term I encountered in Austin, if nowhere else.

> It would be cool to have a bsg package that allows you to panelize your
HW...

I'm not sure how the backend was implemented (except that it was using curses), but essentially how it worked was by making a template, with some special syntax to indicate a Verilog variable vs a local variable.  Roughly:

a[63:0] = VPI("testbench.u1.u2.dout[127:64]")

[`a[0]` | `a[1]` | `a[2]` | `a[3]` | `a[4]` ]
[`a[5]` | `a[6]` | `a[7]` | `a[8]` | `a[9]` ]
[Expression: `(a[7] ^ a[8] | a[9)`]

would lay out a table, with the ability to scroll back and forth through clock cycles.  The bottom row would display the result of that expression. It was nice because it abstracted most of the actually VPI work away and let people build complex visualization structures fairly easily

taylor-bsg, 2019-04-25:
Ah I see! Rough thoughts below:

Putting aside the question of being able to go back in time, which is an
interesting feature, but maybe a little heavy-weight in the general case
(i.e. being able to show contents of memory across all time).

I think since we have variable numbers of cores etc, we might want to keep
things determined programmatically rather than in a closed form text
language, but make it relatively simple.

I think the key abstraction would be a "panel", which has this interface

// panel initialization

for core = 0; core < num_cores; core++ {
(0) an interface for creating a new panel of a particular type:
                                    panel = new MyFunkyPipelinePanel
(1) an interface for mapping verilog paths to particular field numbers and
their data types:  panel->map_field(field number, data_type, verilog_path,
optional_description_string);  e.g. top.foo.core[core].PC
}

// execution

if (input == "s") {
verilog->step();

// panel updating
for each panel, if active {
(2) an interface for updating data information:
                                            panel->update_data();
                        // customized for the type of panel
     by default this would capture the data from VPI, but it could be
swapped if there was something else
(3) an interface for rendering the panel after field updates have been
performed:                 panel->render_fn->render();                //
customized for the type of panel (MyFunkyPanelRender)
                 the render function would probably have a particular class
that defines the difference kinds of rendering backends (with NCURSES
panels being the first one we support)
                // draws fancy pipeline diagram; can post process input
data in C to draw "exactly the right thing"
               // can also have standard ones that dump out standard data,
or take a C string with fields "%s %d %f\n%d %c %s";
(4) an interface for changing what the panel is displaying (For example,
the base address for a DRAM dump.)
  }
}

M

dpetrisko, 2019-08-01:
New BSG project release coming up soon :)

closed 2019-08-01 by dpetrisko