black-parrot/black-parrot #147: Unsupported features for OpenSBI boot

dpetrisko, 2019-04-26:
OpenSBI supports a very limited amount of emulation. We can either hack it to bits or wait until we have this support.

1) Multiply
2) Floating point (hard)
3) Atomics (although I believe we can insert our emulation library here)

taylor-bsg, 2019-04-27:
What determines what OpenSBI does or not support? Is that OSBI the standard
or the implementation?

dpetrisko, 2019-04-27:
OpenSBI is the open-source implementation (https://github.com/riscv/opensbi) of the RISC-V SBI Spec (https://github.com/riscv/riscv-sbi-doc/blob/master/riscv-sbi.md)

taylor-bsg, 2019-04-27:
In what way does it support or not support
emulation? Does it mostly run in m-mode and the make use of those features
in m-mode?

dpetrisko, 2019-04-27:
Does not support may be overstating it.  It runs in M-mode and emulates operations from lower privilege modes.  For these operations, it uses multiply, float and atomics and doesn't provide a mechanism to emulate them yourself.  For floats specifically, it flags a build error if you try to link with softfloat libraries.

I have no doubt given enough hacking that we can get past it, but if we're implementing these features anyway, it's not necessarily worth the effort.  BBL can compile with just RV64I and is also capable of booting Linux

taylor-bsg, 2019-04-27:
The SBI is basically the BIOS right? So we don’t need to be software
compatible?

What does it use floats for?

M

dpetrisko, 2019-04-27:
Tagging in @HelloKayT for the first two questions.

Looking into it, it seems to only deal with managing the float regfile and fcsr, so it's probably possible to just rip out those calls. Doesn't actually use fpu ops for anything

dpetrisko, 2023-07-18:
We have all of this support, so closing (although interesting discussion)

closed 2023-07-18 by dpetrisko