black-parrot/black-parrot #148: Genericize cce-mem interface

dpetrisko, 2019-04-27:
We should genericize the cce-mem interface.  Possibly this should be AXI or something similar.  Otherwise:

1) Rename the cmds without cce_*
2) Change the forwarded payload as raw bits instead of a specific struct (like in FE-BE interface).  This will decouple the CCE implementation from the memory interface.  https://github.com/black-parrot/pre-alpha-release/blob/be_dev/bp_common/src/include/bp_common_me_if.vh#L517
3) Change lce_cce enums to a generic name. For 1 bit enums, possibly change to raw logic for clarity

taylor-bsg, 2019-04-27:
Keep in mind, this is the CCE-L2 interface, not the CCE-DRAM interface...

dpetrisko, 2019-04-27:
sed -i "s/cce_mem/l2_mem/g"

taylor-bsg, 2019-04-27:
I think I am getting the gist of what you are saying —- basically there
should be interface like we have for BE/FE that does not expose unnecessary
inner details of the directory engine; for example in a single core
implementation, we would not need a lot of this.

In a nutshell:

A. we should have something that looks like an L2 interface, not a CCE/L2
interface.
B. Similarly for L2/DRAM.
C. And there should probably be a third interface which is our I/O enclave,
which is just off of the CCE.
D. (And I guess there is a fourth item, which is the coherent I/O
interface, but that should be a lightly modified LCE interface.)

I think the mechanism there is to provide for synthesis time
configurability of a metadata payload, much like we do with the BE/FE.

Regarding AXI, it is not a good choice for our native interfaces — but is a
must-have for our I/O interfaces — we should have adapters in the BP repo
that allow for easy interfacing.

dpetrisko, 2019-04-27:
Makes sense to me.  This is coming up because I'm writing the I/O enclave.  Right now it's over-provisioned and just re-uses the CCE/L2 interface.  Looks like the path forward is to design a more light-weight, non-coherent I/O interface. 

In Ariane, they used Axi-Lite as the CLINT interface https://github.com/pulp-platform/ariane/blob/master/src/clint/clint.sv

Rocket uses TileLink https://github.com/freechipsproject/rocket-chip/blob/master/src/main/scala/devices/tilelink/CLINT.scala

dpetrisko, 2019-05-15:
I ended up doing something very simple for the CLINT interface and punted the I/O decision until later.

https://github.com/black-parrot/pre-alpha-release/blob/dev/bp_top/src/v/bp_clint.v

Essentially, the CCE-MEM interface is the same, but since the timer and software interrupts a) always accept and b) are stored in flops, we just read the request and forward it back with no additional overhead. When we end up handling I/O, (and expand CLINT to I/O enclave), we should use the redesigned interface for outward facing signals.

closed 2019-08-14 by dpetrisko