black-parrot/black-parrot #149: Need to differentiate LR and emulated AMO for store page fault

dpetrisko, 2019-04-28:
RISC-V defines Store/AMO page fault as an exception.  From my understanding, LR should trigger a load page fault.  However, when we emulate an AMO, the sequence is LR-do op-SC. If we page fault on that first LR, we should throw a Store/AMO page fault instead of load page fault.

Probably we should handle this in M-mode software.  We could have a flag which says we're in an AMO emulation routine so that when we trap on a page fault, we check that bit to differentiate.

Or it's possible that since we're in M-mode, software compliance is not that important.

taylor-bsg, 2019-04-29:
Interesting. But isn’t VM turned off in M-mode when we emulate the
instruction? So we would actually be doing the page table walk manually,
 and would just somehow have to vector over to the page fault exception
vector (with associated delegation?) after restoring registers? (Talking
out of my butt...)

M

dpetrisko, 2019-04-29:
Oh, that’s true. I’m realizing that I hadn’t considered all of the implications of emulating AMOs with VM. With a software page table walker, I think it makes sense. We’ll have access to all the CSRs and be aware of the routine that we’re faulting in, so it’s reasonable to expect that to be handed by software. 

I feel more uneasy about emulation if we have to do a page table walk on every atomic, though. 

Here’s a thought. Emulation is triggered by an illegal instruction exception. We could do the translation in hardware while passing down the pipe, and then give the translation to M-mode via an “emulation paddr CSR”. This would solve the fault issue as well as preventing a software PTW

taylor-bsg, 2019-04-29:
This is kind of what I was alluding to when pushing back on mark about
emulation. When you push things to software, then your architectural
interfaces get more complicated because now the things you had access to in
HW have to be exposed in SW. And after you go to the trouble of adding to
the architecture to allow things to be done in SW, we will find out that
performance is bad and go back to HW. Or we will find out that it is
incompatible with SiFive’s attempts to enshrine their silicon into the
spec, or OpenSBI or whatever.

I think the two “fixed points” are all-software emulation (good for “A” in
PPA, and for quickly tracking the functionality of the evolving spec before
we figure out the right way to do it in HW, or have the bandwidth in our
schedule), or, do it in HW. We already need the support for executing
sequences of instructions atomically, for LR/SC, which presumably we can
layer on top of.

I think for the software case, the TLB walk is not that bad considering we
are already saving registers etc.

dpetrisko, 2019-04-29:
That’s fair, although I think there could be more nuance in accelerated emulation potentially being a P3/P4 design, especially if the extra hardware is shared among all emulated features. Probably need to have a larger discussion about emulation vs hardware for various things now that we’re at the point of implementation. 

My concern is with maintaining a software PTW in addition to the hardware PTW, with all the accessed / dirty bit handling that implies, in addition to the aforementioned complications with fault handling. I think an extra 64 bit register is a fair trade-off for that

taylor-bsg, 2019-04-29:
We should definitely hold off on prematurely optimizing the emulation path
until we can get look across the emulation code and identify the
commonality and the hot paths and can make a more educated determination of
what should be emulated and what should not... before you add hardware you
should always have the software equivalent so you can measure the benefit
and make sure it is as much as you think.

Aside: One thing I was realizing as well, is that given the nebulous nature
of RISC-V, we also need to consider the possibility that if we are
successful, we will need to be backwards compatible with ourselves.  The
problem with adding architectural registers to implement
micro-architectural features is that those things then have to be supported
even when the micro-architectural motivation for those things ends.

M

dpetrisko, 2019-05-06:
Katie reminded me that there's a bit in mstatus CSR which allows you to have translation enabled in M-mode, so we don't need a software PTW.

taylor-bsg, 2019-05-07:
To clarify, you mean; a software PTW will not work unless we have a level
below M-mode? Makes sense to me.

dpetrisko, 2019-05-07:
I'm saying a software PTW is unnecessary. Once we enable paging, we can set mstatus.MPRV. MPRV means that we treat our D$ VM translation as if we were in the previous privilege mode (S or U). When we trap into M-mode to perform the emulated AMO with full VM support.

taylor-bsg, 2019-05-15:
Got it! I was thinking of some corner cases where we have TLB misses in the
middle of an AMO, causing a livelock, but I think it should be generally
okay, since the TLB miss will happen before the LR completes.

M

dpetrisko, 2019-05-15:
Hm.

1) a) LR causes PF. 
    b) Take horizontal trap
    c) service
    d) replay LR
2) a) LR succeeds
     b) SC causes PF. 
     c) Take horizontal trap & reservation is cleared
     d) Replay SC (fail)
     e) Go to a

I guess there could be a loop if some other core is repeatedly doing TLB shootdowns, but I don't think that's a case we should worry about

dpetrisko, 2020-11-21:
I believe this is resolved as "not a bug", but possible, unlikely attack vector

closed 2020-11-21 by dpetrisko