black-parrot/black-parrot #154: Need to flag error if running non-existent program

dpetrisko, 2019-05-07:
Right now, when you run 

`make run.v PROG=my_prog`

it copies the test program from bp_common into run_vcs and then . runs the simv using that program.  However, if you do `make run.v PROG=myporg` and it can't find the correct program, the copy will fail silently and use the old program, sometimes resulting in a spurious PASS.

taylor-bsg, 2019-05-07:
I don't know the context, but generally, the correct way to do this is to
use a different directory for each different run you do. One directory, one
set of expected results. This also allows you aggregate results, summarize
them, and prevent these kinds of problems.

m

dpetrisko, 2019-05-21:
@taylor-bsg @stdavids 
Is this a good directory structure for this?
I was unsure where the simv should go, since we build once and then use that simv for all test programs.

reports/<tool>/<tb>.build.rpt
reports/<tool>/<tb>.build.err
reports/<tool>/<tb>.sim.<prog>.rpt
reports/<tool>/<tb>.sim.<prog>.err

logs/<tool>/<tb>.build.log
logs/<tool>/<tb>.sim.<prog>.log

run/<tool>/<tb>.build/simv
run/<tool>/<tb>.sim.<prog>/(symlink to simv)
run/<tool>/<tb>.sim.<prog>/dump.vcd
run/<tool>/<tb>.sim.<prog>/testbench.v
run/<tool>/<tb>.sim.<prog>/etc.

stdavids, 2019-05-21:
I like it! Maybe consider renaming "run" to "results"? That is more consistent with our other cad infrastructure... soft suggestion though. The other thing is that simv is tied with a particular cfg_p right? So maybe the generated simv should be changed to simv.\<cfg_p\>?

dpetrisko, 2019-05-21:
Oh, good point! Probably all of the `<tb>` should then become `<tb>.<cfg>`, including the simv directory

"results" works for me. I can see why it would make more sense in the context of say, DC

dpetrisko, 2019-05-26:
Live in dev

closed 2019-05-26 by dpetrisko