black-parrot/black-parrot #157: Floating point should be made optional

dpetrisko, 2019-05-09:
Right now, we have a floating point register, forwarding paths, pipeline stages, etc. despite not supporting hardware float. But, we don't want to outright remove it because we will eventually support hard float. Therefore, we need a parameter which enables or disables FP things in the core. The trick is to do this without making the BE more fragile or less readable.

taylor-bsg, 2019-05-09:
Can it go in its own module?

M

dpetrisko, 2019-05-09:
The changes are pretty dispersed throughout the Calculator. I’ll have to think about it. Possibly if we encapsulate regfiles + bypass in a single unit, that could be a good partitioning for one piece

dpetrisko, 2019-05-15:
Live in dev. There are some extra optimizations that could be made to e.g. shorten the pipeline length. But we might extend it for the iterative multiplier anyway, so it seem premature to mess around with it

closed 2019-05-15 by dpetrisko