black-parrot/black-parrot #158: Fix DRAMSim2 for Verilator

dpetrisko, 2019-05-09:
Our Verilator testbench stalls on during DRAMSim2 DPI load calls (although it reads instructions fine). This is blocking full-system Verilator testbenches from working.

dpetrisko, 2019-05-16:
Update: I've traced the issue down to the wormhole router.  It may be the network connection, but the link works fine in VCS.  AFAIK, the wormhole router has never been tested in Verilator, so there may be some minor incompatibility there.

dpetrisko, 2019-05-16:
Fixed pending a change in Basejump STL noc_pkg.

```
// direction type
package bsg_noc_pkg;
  typedef enum logic[2:0] {P=0, W, E, N, S} Dirs;
endpackage

`endif
...
  localparam LEN = 3;
  localparam COL = 4;
  localparam SEQ = {LEN'(N), LEN'(E), LEN'(W), LEN'(P)
                   ,LEN'(S), LEN'(E), LEN'(W), LEN'(P)
                   ,LEN'(S), LEN'(N), LEN'(W), LEN'(P)
                   ,LEN'(S), LEN'(N), LEN'(E), LEN'(P)
                   ,LEN'(S), LEN'(N), LEN'(E), LEN'(W)};
```
Would come up as a concatenation of unsized variables, breaking the logic.  Fix is to change this line.

```
  typedef enum logic[2:0] {P=3'd0, W=3'd1, E=3'd2, N=3'd3, S=3'd4} Dirs;
```

taylor-bsg, 2019-05-16:
Crazy... seems like a Verilator bug? If so, annotate line with Verilator
version number

dpetrisko, 2019-05-16:
In here or in bsg_noc_pkg?

taylor-bsg, 2019-05-16:
The latter

dpetrisko, 2019-05-16:
I'll add the note, but it seems like best practice should be to explicitly size values, anyway

taylor-bsg, 2019-05-16:
Interesting but the type is 3-bits so why is this necessary? Do you have a
link somewhere?

dpetrisko, 2019-05-16:
Here's the bug report I submitted.

https://www.veripool.org/issues/1442-Verilator-Enum-values-without-explicit-widths-are-considered-unsized

I don't think it's strictly necessary per IEEE. I think what should happen is the value gets truncated in the typedef and then is correctly sized from then on. What does happen is probably Verilator stores the enum value as symbol without considering the size.  So when you do {E, E, W} it's replacing it with {1, 1, 2}

closed 2019-05-16 by dpetrisko