black-parrot/black-parrot #163: Enable -j for building toolchains

dpetrisko, 2019-05-13:
Building the toolchains take a long time. Right now, we use recursive makefiles to build the submodules. AFAICT, passing -j from the top level makefile does not pass to the lower level makefiles. I'm sure there's some easy makefile syntax to enable this

stdavids, 2019-05-13:
If you use "$(MAKE)" instead of "make" I believe it will pass down the makefile flags. See below for an example:

[image]

dpetrisko, 2019-05-13:
What's with the 
warning: jobserver unavailable?

taylor-bsg, 2019-05-13:
Not sure this the right behavior — do you want 4X parallelism
multiplicatively at every level of the hierarchy? Better to build it into
the submakefiles so everybody runs the same ways and has the same
expectations of how safe the makefile are.

M

dpetrisko, 2019-05-13:
As a user I think I'd like to do  `make tools -j 4` and have that do (in serial)
`make gnu -j 4`
`make verilator -j 4`
`make systemc -j 4`

taylor-bsg, 2019-05-14:
Seems like gmake does support doing a max of  N threads across recursive
makefile s:

The ‘-j’ option is a special case (see section Parallel Execution
<https://www.tack.ch/gnu/make-3.82/make_40.html#Parallel>). If you set it
to some numeric value ‘N’ and your operating system supports it (most any
UNIX system will; others typically won’t), the parent make and all the sub-
makes will communicate to ensure that there are only ‘N’ jobs running at
the same time between them all. Note that any job that is marked recursive
(see section Instead of Executing Recipes
<https://www.tack.ch/gnu/make-3.82/make_81.html#Instead-of-Execution>)
doesn’t count against the total jobs (otherwise we could get ‘N’ sub-makes
running and have no slots left over for any real work!)

dpetrisko, 2019-07-25:
Fixed in dev

closed 2019-07-25 by dpetrisko