black-parrot/black-parrot #169: Should dynamically set max simulation time based on test

dpetrisko, 2019-05-17:
...or stop CI on first failed test.  Otherwise, failed regressions are seriously going to eat into our regression time.  The latter is easier to implement, so I'll do that first as a stopgap.  We can't lower the max sim time in general because we have relatively long-running benchmarks

taylor-bsg, 2019-05-17:
Typically we put in a default amount of time and then allow that to be
overridden in the Makefile. When people get the time, they will figure out
tighter bounds and put them in.

M

dpetrisko, 2019-05-17:
Yeah that sounds good. But we need to be static for CI. Probably just a json that the CI reads and overrides the simtime from (defaulting if it’s not found)

dpetrisko, 2020-04-26:
This is less important with the watchdog timer. I'll keep it open since we should also revamp the sim flow a bit

dpetrisko, 2023-07-18:
Closing as not useful

closed 2023-07-18 by dpetrisko