black-parrot/black-parrot #173: Deduplicate network code

taylor-bsg, 2019-05-19:
https://github.com/black-parrot/pre-alpha-release/blob/db044930c35844e69b2d3367e510c4562af50e29/bp_top/src/v/bp_tile.v#L247

Needs some refactoring;  see these links for how to organize the code:

https://github.com/bespoke-silicon-group/bsg_manycore/blob/master/v/bsg_manycore.v

https://github.com/bespoke-silicon-group/bsg_manycore/blob/master/v/bsg_manycore_tile.v

https://github.com/bespoke-silicon-group/bsg_manycore/blob/master/v/bsg_manycore_mesh_node.v

dpetrisko, 2019-05-19:
Is anyone else using the wormhole router? I would like to refactor it to use the bsg noc links, which would help with this.

taylor-bsg, 2019-05-19:
Paul and Farzam will be using it for chip-chip I/O. Since Paul is the
author, I suggest you give him the feedback and he can make the changes;
and you can make the parallel changes to BlackParrot.

dpetrisko, 2019-05-19:
Sounds good! Discussion moved to https://github.com/bespoke-silicon-group/basejump_stl/issues/43

dpetrisko, 2019-09-11:
Fixed in top_dev