black-parrot/black-parrot #182: CCE rom should be binaries, not ROMs

dpetrisko, 2019-05-25:
What we have now:

1. CCE roms are prebuilt for some crossproduct of LCE/CCE/sets
2. We select a bp_cfg from Makefile
3. Using a static mapping in the makefile, we cp in the rom to the build directory and add it to the filelist
4. We compile the rom and use it as input to the config loader

What we should do:
1. Prebuild the CCE binaries
2. We select a bp_cfg from Makefile
3. cfg loader reads that config and loads the CCE binary from readmemh

There's also a "next step" about having the CCE calling a shell script to build the rom just in time, but that may bring up some platform dependency issues whereas this solution does not.

BONUS: We can get rid of tons of CCE roms clogging up our git repo.

muwyse, 2019-05-28:
Can parameters from the verilog module be used to generate the file name string for readmemh? E.g., lce_assoc_p = 8 --> readmemh("rom_name_assoc{lce_assoc_p}")

Is this what the double tick notation `` can do?

muwyse, 2019-05-28:
I can pretty easily create binaries instead of ROMs, but I don't think this will get rid of the large diffs. Anytime the microcode changes, the binaries need to be rebuilt. Since the input to readmemh is a .mem file, which is really a text file with a bunch of hex, won't the git diffs still show this?

dpetrisko, 2019-05-28:
> Can parameters from the verilog module be used to generate the file name string for readmemh?

You can do sformatf, like in C, with either parameters or logic.
https://github.com/black-parrot/pre-alpha-release/blob/dev/bp_be/test/common/bp_be_nonsynth_tracer.v#L103

> Since the input to readmemh is a .mem file, which is really a text file with a bunch of hex, won't the git diffs still show this?

We wouldn't commit the binaries

dpetrisko, 2019-09-11:
Fixed at some point

closed 2019-09-11 by dpetrisko