black-parrot/black-parrot #184: Some description of what clint is and does.

taylor-bsg, 2019-05-25:
https://github.com/black-parrot/pre-alpha-release/blob/548b0b6fc993124406cd7270bf9ceb8877ed8be6/bp_top/src/v/bp_clint.v#L1

dpetrisko, 2019-05-26:
The CLINT (core level interrupt controller) handles local (non-external, i.e. not device) interrupts.  For RISC-V there are two types: timer and software interrupts.

Timer interrupts are controlled by two memory mapped registers. mtime is an RTC (we just implement as a simple counter), shared among all cores. mtimecmp is per core. When mtime >= mtimecmp, a timer interrupt is generated and sent to the the core.

Software interrupts are controlled by a memory mapped register per core. Any core in M-mode may write to any other core's software interrupt pending register.

Originally, the CLINT design was 1 per SoC. It snooped the CCE dram connection, checks for matches and then sent the single bit interrupt signals to other cores.  However, since the dram connection will be wormhole routed, it makes sense to put the CLINT on an I/O router. 

MTIME and MTIMECMP should be directed over the network as normal memory requests with the CLINT as a destination.

Software interrupts should be directed to the CLINT. The CLINT will then send an ack to the sending core and a memory cmd into the destination tile (this requires an additional bit for interrupt type message).

taylor-bsg, 2019-05-26:
Add as comment to the source file?

Clint snooping all memory ports: non-scalability eek.

Random question — how are uncached I/O accesses routed to directory shards? Presumably they should always go to the local shard — gives parallelism and maintains SC for the same endpoint.

How are software interrupts represented at the target core? As a bit vector?

What do you mean by “the Clint will send a memory cmd into the destination tile”?  Which CCE network does it travel on?

dpetrisko, 2019-05-26:
Yeah, I’m going to rewrite to hang off the NOC, so it’ll only be 1 memory port. This version was more to test the interrupt functionality before we implemented the NOC. 

I’m not sure I understand the second, @muwyse ?

Software interrupts are a single bit. They’re cleared by writing to that same memory mapped address, except that a core can write its own SIP in S or U mode (whereas M mode can write to any core). 

Right now there are just three wires (time/soft/ext int) going into the tile which feeds into the interrupt pending bits of the mstatus CSR. I figured that interrupts don’t need consistency, so they don’t travel over an LCE-CCE network. You may disagree :)

taylor-bsg, 2019-05-26:
Clarifying a few things ---

- only MSIP is memory-mapped
- to implement MSIP coherently, when the clint receives a write to the MSIP
location, it should be translated into two bits going over two wires to the
relevant BP core:  0X do nothing, 10 write a 0, 11 write a 1? it is up to
the core to aggregate these writes with local writes being performed by the
core to the MSIP register. The wires should be pipelined to address the
delay across the chip.
- the standard way to set MSIP remotely appears to be with a atomic set bit
instruction, so the clint would have to support atomic uncached stores
(much as we described them working in the manycore; you just send the
ATOMIC request in the packet and the endpoint hardware handles it):

https://github.com/riscv/opensbi/blob/master/lib/sbi_ipi.c
https://github.com/riscv/opensbi/blob/master/lib/riscv_atomic.c

dpetrisko, 2019-05-26:
> only MSIP is memory-mapped

MSIP, MTIME and MTIMECMP are all memory-mapped. MTIP is not memory mapped, because it is controlled by the expression (MTIME >= MTIMECMP)

> to implement MSIP coherently, when the clint receives a write to the MSIP
> location, it should be translated into two bits going over two wires to the
> relevant BP core:  0X do nothing, 10 write a 0, [...]

I think you only need one wire carrying the current state of the bit.  Because the only way to control these CSRs is through the memory-mapped address, from the core's perspective it's simply taking another few clock cycles for the write to propagate.  (agree on the pipelining)

> the standard way to set MSIP remotely appears to be with a atomic set bit
instruction you just send the ATOMIC request in the packet and the endpoint hardware handles it):

Makes sense to me

dpetrisko, 2019-05-26:
One minor detail about pipelining the MSIP write is that we'll need to wait N cycles before sending the RESP on the network, where N is the depth of the pipeline. Otherwise we could end up reading before the write has completed.

taylor-bsg, 2019-05-26:
> only MSIP is memory-mapped
>
> MSIP, MTIME and MTIMECMP are all memory-mapped. MTIP is not memory mapped,
> because it is controlled by the expression (MTIME >= MTIMECMP)
Yes, I was replying about the interrupts -- I mean more precisely,
SSIP/USIP are not memory-mapped

to implement MSIP coherently, when the clint receives a write to the MSIP
> location, it should be translated into two bits going over two wires to the
> relevant BP core: 0X do nothing, 10 write a 0, 11 write a 1? it is up to
> the core to aggregate these writes with local [...]
Okay, I see, the value of MSIP is read-only when accessed directly from the
status register by the local core; so we don't need to worry about
simultaneous writes. Just incoherent reads. =)

riscv-privileged-20190405: The mip register is an MXLEN-bit read/write
register containing information on pending interrupts, while mie is the
corresponding MXLEN-bit read/write register containing interrupt enable
bits. Only the bits corresponding to lower-privilege software interrupts
(USIP, SSIP), timer interrupts (UTIP, STIP), and external interrupts (UEIP,
SEIP) in mip are writable through this CSR address; the remaining bits are
read-only.

mip register:
WPRI, MEIP, WPRI, SEIP, UEIP, MTIP, WPRI, STIP, UTIP, MSIP, WPRI, SSIP, USIP

dpetrisko, 2019-05-26:
The only question I have is how to address the CLINT.  The simplest way would be that the wormhole adapter will have a static routing table with the MMIO addresses to set the dest coordinates. I think this would probably scale for extra I/O devices.  Separately, we need the LCE to recognize these addresses as uncacheable.

Having a general range comparison in LCE and then specific == comparison in WHA seems reasonable to me.

taylor-bsg, 2019-05-26:
a. actually I don't think they do use an atomic bit set on the ipi address

b. I emailed the mailing list to clarify the semantics of the two ways to
access MSIP

M

>

taylor-bsg, 2019-05-26:
Agree -- (PA -> cacheable / uncacheable) in LCE, and (PA -> NPA {network
physical address == X,Y on wormhole network}) as a message passes into the
wormhole adapter (maybe not part of the NOC->wormhole code itself, since
this would not be very generic.).

dpetrisko, 2019-05-26:
Ah from Andrew’s response, it looks like the spec is pretty unrestrictive on the timing. We can do the pipelined wires without worrying about synchronization

taylor-bsg, 2019-05-26:
Indeed!

The relevant email chain:

https://groups.google.com/a/groups.riscv.org/forum/#!topic/hw-dev/WmmnEPy4X2E

dpetrisko, 2019-05-27:
https://github.com/black-parrot/pre-alpha-release/blob/top_dev/bp_top/src/v/bp_clint.v

Rework.  Paul's fixing up the CCE-MEM interface so it'll change a little bit.

dpetrisko, 2019-05-30:
I'll leave this open with an action item 

- [ ] Write up a doc that describes the MMIO enclave, including the memory-map for the CLINT registers, cfg link bus, etc.
- [ ] Describe the interaction between memory-mapped R/W, CSRRW, and irq pipelines
- [ ] Identify all potential races

muwyse, 2019-05-30:
>  Paul's fixing up the CCE-MEM interface so it'll change a little bit.

What exactly is changing here? If it is changes to the structs or what the CCE sees, this could require changes in the CCE, too.

dpetrisko, 2019-05-30:
You might want to ask him, but I looked at the bp_common/me_if and nothing has changed.  I think he might have meant adding the wormhole header, but that doesn't change the actual interface once the packet is extracted.