black-parrot/black-parrot #187: Should be a decoder in BaseJump STL you can use.

taylor-bsg, 2019-05-27:
https://github.com/black-parrot/pre-alpha-release/blob/19cd8defcda72d27de7595a615c4e3c49a01869c/bp_top/src/v/bp_clint.v#L58

dpetrisko, 2019-05-27:
Are you saying there is one, or I should I write one?

taylor-bsg, 2019-05-27:
https://github.com/bespoke-silicon-group/basejump_stl/blob/master/bsg_misc/bsg_decode_with_v.v

https://github.com/bespoke-silicon-group/basejump_stl/blob/master/bsg_misc/bsg_decode.v

dpetrisko, 2019-05-27:
Oh, I somewhat see what you mean.

Based on other feedback, I moved the address comparison inside of the generate:
https://github.com/black-parrot/pre-alpha-release/blob/top_dev_feature_clintpdrework/bp_top/src/v/bp_clint.v

This compares the full address.

The other option is to decode_with_v  and have v be (mem_data_cmd_v & a comparison with the base address).  This probably saves on comparators

taylor-bsg, 2019-05-27:
Yep!

taylor-bsg, 2019-05-27:
Yes I suggest switching to decode_with_v as you mentioned.

dpetrisko, 2019-05-27:
Fixed

closed 2019-05-27 by dpetrisko