black-parrot/black-parrot #189: Clint write only?

taylor-bsg, 2019-05-27:
https://github.com/black-parrot/pre-alpha-release/blob/19cd8defcda72d27de7595a615c4e3c49a01869c/bp_top/src/v/bp_clint.v#L156

I believe that these registers need to be readable as well per the spec.

dpetrisko, 2019-05-27:
Where in the spec do you see this? I don't see anything mentioning memory-mapped reads

> Platforms provide a real-time counter, exposed as a memory-mapped machine-mode register, mtime. mtime must run at constant frequency, and the platform must provide a mechanism for determining the timebase of mtime.

> The machine-level MSIP bits are written by accesses to memory- mapped control registers, which are used by remote harts to provide machine-mode interprocessor interrupts.

dpetrisko, 2019-05-27:
Also, if mtime needs to be writable, then we'll need to modify bsg_cycle_counter to support setting.

taylor-bsg, 2019-05-27:
Seems like there should be one read/writable mtime and mtimecmp register
per core? Maybe it does not make sense to put it on the network?

M

dpetrisko, 2019-05-27:
mtime should be shared among all cores. If it's an RTC we only want to have 1 per system. Then, mtimecmps should live near it so we're only sending the timer_irq across the proc instead of the mtime value.

(oh, and mtime should definitely be readable. whoops, I was mixing up a few requirements).

taylor-bsg, 2019-05-27:
If you read the spec carefully, I think they say you need a real time clock
that is uniform across the system, but that each core had its own
read/writeable counter?

M

dpetrisko, 2019-05-27:
The spec is ambiguous, but Rocket and Ariane both only have 1 mtime register and N mtimecmp registers.  Back when mtime and mtimecmp were CSRs, I believe it was 1 and N as well (although I can't find an old enough spec to prove it).

On the one hand, it makes sense that having mtime be global means that you could sync timers between cores more easily. On the other, you don't want cores to clobber each other's timer interrupts, which is why I figured read-only mtime makes sense.

dpetrisko, 2019-05-27:
https://github.com/black-parrot/pre-alpha-release/blob/top_dev_feature_clintpdrework/bp_top/src/v/bp_clint.v

I made them all (including mtime) MMIO read/write capable for now.  We can take out pathways when we confirm either way

taylor-bsg, 2019-05-28:
Great — seems like they should all be read/write. The RISC-V forum has a
couple of code snippets where mtimecmp is read.

M

taylor-bsg, 2019-05-28:
Just putting this here for reference.

https://groups.google.com/a/groups.riscv.org/forum/m/#!topic/hw-dev/m3VoPhteSgw

dpetrisko, 2019-05-30:
Fixed on off-chip network branch. Will get merged in with those changes

closed 2019-05-30 by dpetrisko