black-parrot/black-parrot #193: clock discipline violation

taylor-bsg, 2019-05-29:
https://github.com/black-parrot/pre-alpha-release/blob/19cd8defcda72d27de7595a615c4e3c49a01869c/bp_top/src/v/bp_clint.v#L117

This is an incorrect clock domain crossing. There are two fixes:

1) remove rtc_i and use clk_i. this means our RTC is full clock speed.

2) obey clock discipline
    a) per the systemverilog coding guidelines, label every signal in the module with what clock it belongs to. 
    b) figure out where you will cross clock domains.
    c) if it is single wire signals crossing, then you can just use bsg_launch_sync_sync. if it is multiple wires, then it gets very complicated. you can either use an bsg_async_fifo; or a simpler async handoff module (which we have not put into basejump_stl yet, but should; we have several versions of it floating around our code base.) Most likely you would chose to put the comparators in the same clock domain as the RTC, and have the interrupt lines come out; but you would still need to implemented reading/writing across the domain.

I suggest we apply fix #1 for the tapeout, and fix #2 afterwards.

dpetrisko, 2019-05-29:
1. Works for me.

dpetrisko, 2019-05-30:
1. applied to off-chip-network branch, will get merged in with that

dpetrisko, 2019-09-26:
> if it is single wire signals crossing, then you can just use bsg_launch_sync_sync. if it is multiple wires, then it gets very complicated. you can either use an bsg_async_fifo; or a simpler async [...]

https://github.com/pulp-platform/ariane/blob/master/src/clint/clint.sv#L158
Ariane has the RTC increment crossing, which is certainly the lowest hardware solution.  It will glitch occasionally of course, but possibly that's acceptable.

>(which we have not put into basejump_stl yet, but should; we have several versions of it floating around our code base.) 

Could you send a pointer? If they're lighter weight than a FIFO, we should be using them for our async mem crossings

taylor-bsg, 2019-09-26:
Good point you could use BaseJump STL to grey code the RTC. This way only one bit changes at the sender, so when it is latched, there is only one bit that is potentially metastable, and it doesn't matter which way it goes -- you either get the previous value or the next value.

What that said, if you can just send the increment signal across and you know the receive clock is much faster, that is even lighter weight.

dpetrisko, 2021-01-12:
3 possible modes for RTC
- Cycle ratio (dynamic?)
- Instruction retirement (synth or sim only?)
- Actual RTC (crossing strategies discussed above?)

omasanori, 2022-02-28:
Was this addressed by https://github.com/black-parrot/black-parrot/pull/1019?

dpetrisko, 2022-02-28:
Yeah, we can close this when we merge:
https://github.com/black-parrot/black-parrot/pull/1047

(pending a minor CI fix)

closed 2022-02-28 by dpetrisko

omasanori, 2022-02-28:
I see, so merging to `dev` is criteria for completing tasks, is it? Thank you for your response, @dpetrisko!

dpetrisko, 2022-02-28:
Yep, dev is considered mainline. <end>_dev are internal development branches.  master happens after we do ASIC and FPGA evaluations, so it lags a bit but is more stable

omasanori, 2022-02-28:
OK, thanks!