black-parrot/black-parrot #196: Stall timeout for BE

dpetrisko, 2019-05-30:
We should add a stall timeout in the BE which says that we haven't got a fetch in N cycles, something must be wrong in the FE, send a redirect to revive it.

This should print an assert so that we can see in simulation that a bad condition was reached, but it will allow us to make progress regardless.

taylor-bsg, 2019-05-30:
Super interesting (but super appalling) idea, especially at the meta level.
You guys make me question all of my values, it is fun.

I know in the Pentium 4 they had cases where the machine could deadlock,
and had to add a similar “defibrillation” mechanism. In an Alpha 21264, the
register renamer would leak registers off the free list and the machine
would slow to a crawl.

But our machine is not so complicated — if we can’t prove that the front
end doesn’t do this, then we should rewrite the code.

Before we tapeout, we will have run whatever programs we expect to work
through simulation. They will work. Everything else is a freebie. If the
chip has some weird lockup bug, we want it to lock up, so we know about it
and can fix it in the next tapeout. The relevant Taylor’s Laws:

- Kill bugs, don’t chase them away. (A bug sighting is an opportunity not
to be squandered...)

- Fail fast.

So what is the issue with front end defribillation?  How would we ever know
what the maximum number of stall cycles is, with multi-core, FPGA memory
system, etc? How would we test that the defibrillation results in a
consistent machine state? How do we know that it actually is triggering,
but only working by accident, and would not work after we make some minor
change? Could we just use the timer interrupt to accomplish this?

So I think takeaways — a) a bind statement that detects if the processor is
not committing instructions after 1 million cycles of simulation and
terminates the simulation with some information (is there any instruction
that blocks indefinitely, waiting for an event?) b) some non-invasive way
to output a fe redirect  stream across the wormhole router bus, if we ran
the BlackParrot complex at a sufficiently low frequency? This would allow
us to know what code path code us somewhere. Don’t think we have time for
this on this tapeout, but seems useful.

M

dpetrisko, 2019-05-30:
> Super interesting (but super appalling) idea, especially at the meta level.

My specialty.

> But our machine is not so complicated — if we can’t prove that the frontend doesn’t do this, then we should rewrite the code.

Fair! The only 2 invariants that the FE-BE connection have is that the pc-instr pairs are correct and that the FE will accept a cmd per cycle. But, the FE goes into a wait state when it sends a TLB fill request, possibly speculative. The FE gets out of that wait state by a) receiving the fill response or b) receiving a redirect.  Intuitively, if the FE is sending an instruction stream that contains a TLB fill request, the only two options are for the BE to reject an instruction preceding it or return the fill request. So I suppose with the precondition of having no bugs, the FE cannot stall forever unless the ME stalls forever.  

> So what is the issue with front end defribillation?  How would we ever know what the maximum number of stall cycles is, with multi-core, FPGA memory system, etc? How would we test that the [...]

I won't argue against the individual points because I agree overall. However, this does bring up a point about chip debugging.  A timer interrupt will change the machine state before dumping, but assuming you trust the exception mechanism, you could dump the state of the processor to some MMIO address.  Of course, there's some software complexity of having to multiplex actual timer interrupts with this deadlock timer.  

Just spitballing, but how about:

- A watchdog timer triggering an external interrupt. (It can live in the CLINT for now, but should be attached to the PLIC when we have it).
- That interrupt handler just dumps (prints to console) the state of the processor before the interrupt and halts

> b) some non-invasive way to output a fe redirect  stream across the wormhole router bus, if we ran the BlackParrot complex at a sufficiently low frequency?

Interesting idea.  I was thinking earlier about some crazy thing like a circular event tracer RAM that gets written by cores as they commit instructions. Then the host controller can freeze the processor and read from this RAM to see the last N instructions committed and associated state. Kind of like a hardware debug log.

> a) a bind statement that detects if the processor is not committing instructions after 1 million cycles of simulation and terminates the simulation with some information (is there any instruction that blocks indefinitely, waiting for an event?)

Reasonable compromise.

dpetrisko, 2019-05-30:
Long term, how do we feel about (a subset of) the riscv-debug-spec?

https://github.com/riscv/riscv-debug-spec/blob/master/riscv-debug-draft.pdf

taylor-bsg, 2019-05-30:
> Super interesting (but super appalling) idea, especially at the meta level.
>
> My specialty.
>
> But our machine is not so complicated — if we can’t prove that the
> frontend doesn’t do this, then we [...]
This is basically an extra CYCLECMP register  hooked up to an interrupt
line? If we have a PLIC-like thing integrated into the CLINT node, then we
can prototype this and other system architecture features in FPGA to bring
in our tapeout timeline? For the PLIC, does it suffice just to have some
memory mapped registers in the CLINT that tie to interrupt lines that the
FPGA can write to?

> -
>    - That interrupt handler just dumps (prints to console) the state of
>    the processor before the interrupt and halts
>
> b) some non-invasive way to output a fe redirect stream across the
> wormhole [...]

Alpha 21364 had a mode where you could allocate part of the L3 cache to do
this. I think the real question may be, how do you know what to trigger on?
Streaming it off of the wormhole network and taking advantage of the
separate clock domains to reduce interference (I.e. run BP at 1 MHz and the
wormhole link at 500 MHz) is nice because you could get the trace from the
beginning.

But we can also consider the F1 story. We can build F1 images that provide
complete traces for each tapeout, and say because it is open source, you
can run your code on their and get a complete trace.

a) a bind statement that detects if the processor is not committing

dpetrisko, 2019-05-30:
> This is basically an extra CYCLECMP register  hooked up to an interrupt
> line? 

Yes, but the the caveat being that the cycle counter should be reset on a committed instruction.

> If we have a PLIC-like thing integrated into the CLINT node, then we
> can prototype this and other system architecture features in FPGA to bring
> in our tapeout timeline? For the PLIC, does it suffice [...]

Yes, I think what can be feasibly done is have a (host) software writable external interrupt per core, living in the CLINT.  No claim arbitration, no atomics, just devices emulated by the host.  A watchdog timer could live on the FPGA with the caveat that we reset the timer on a DDR request rather than a core level signal.  Come to think of it, that might be cleaner, anyway.

> Alpha 21364 had a mode where you could allocate part of the L3 cache to do this.

Neat!

> I think the real question may be, how do you know what to trigger on?

The long term solution is to set up hardware debug triggers.  I suppose a memory-mapped ->host store wouldn't work well, because you'll lose traces while the store is in-flight.  

> Streaming it off of the wormhole network and taking advantage of the separate clock domains to reduce interference (I.e. run BP at 1 MHz and the wormhole link at 500 MHz) is nice because you could get the trace from the beginning.

My initial reaction is that your program will change characteristics by running at an ultra-slow frequency because of the compute/memory latency ratio. However, I guess that wouldn't change the functional trace in a single-issue, in-order pipe since there's only 1 outstanding memory access at a time (except for multi-core, but non-determinism in multi-core tracing is a whole other can of worms).

> But we can also consider the F1 story. We can build F1 images that provide complete traces for each tapeout, and say because it is open source, you can run your code on their and get a complete trace.

That would be a super cool feature!

taylor-bsg, 2019-05-30:
> This is basically an extra CYCLECMP register hooked up to an interrupt
> line?
>
> Yes, but the the caveat being that the cycle counter should be reset on a
> committed instruction.
>
> If we have a PLIC-like [...]

Sounds nice.

Alpha 21364 had a mode where you could allocate part of the L3 cache to do

dpetrisko, 2019-06-02:
Looks like PULP has a similar idea:

https://github.com/pulp-platform/trace_debugger

taylor-bsg, 2019-06-02:
Nice find! GMTA

dpetrisko, 2023-07-18:
Good discussion; I think all features have been implemented

closed 2023-07-18 by dpetrisko