black-parrot/black-parrot #198: Consolidate and clarify testbenches

dpetrisko, 2019-06-08:
We need to define a testing strategy which provides a high value : maintenance ratio. We need to carefully consider what tests we will maintain going into the future, what tests are required for regression and what tests are needed before a feature is implemented.  Here are some thoughts to seed the discussion.  

My general philosophy on this is that tests are extremely valuable and testbenches are extremely difficult to maintain.  Therefore, we should focus on creating a small set of relatively isolated testbenches which allow a large number of flexible tests to be run.

Testing tools:
- VCS
- Verilator

Testing abstraction levels:
- Top level testing
- End level testing
- Module level testing

Testing strategies:
- Program-based tests
- Random tests
- Directed tests

All of our top-level tests need to run in VCS, because that's what we're using for post-synth testing.  Using the test_bp.v/test_bp.cpp/testbench.v methodology (https://github.com/black-parrot/pre-alpha-release/tree/master/bp_top/test/tb/bp_top_trace_demo) seems to be the easiest way to support both Verilator and VCS for relatively cheap.  Module level tests that run only in Verilator are OK, but ideally should be portable to both. I believe with some care, we can use SystemC for drivers in both VCS and Verilator, although current ME tests do not support this.

We have 1 top level testbench for bp_top and 1 for bp_soc.  bp_top will only have the cores with the top port being the CCE outputs. bp_soc will have the I/O nocs / CDCs, etc.  The DRAMSim2-based fake DRAM will provide program loading and we'll use log-based trace-diffing to verify correctness.  These will support program-based testing and will serve as the most important regressions.

The End level is a bit trickier, because dividing between LCE and CCE is hard, since we don't have a golden model of the CCE. One approach to this is to write a C or Python-based CCE ucode interpreter. This would require maintenance of the interpreter, but would be very nice from the FE and BE perspective, as well as possibly using the golden model to verify behavior of the CCE itself. Another approach is to use a fake cache model, which has the benefit of restricting all tests to the End itself. A third approach is to use the real CCE, which is easiest but breaks isolation between the Ends and so is unideal.  We should use a trace-replay approach for this testing, as it supports both random and directed testing.  Trace-replay can be used to easily generate specific combinations of events e.g. cache miss, branch mispredict and interrupt at the same time.  We can also use Python to generate random combinations of events.

Module level testing is the highest maintenance, lowest reward type of testing.  These tests should be brought up when designing the module but need hard justification for continued maintenance.  At a minimum, the interfaces to the module must be extremely unlikely to change.  Trace replay is not necessary, but suggested at this level.

All maintained testbenches should be on the regression at least in a build capacity.

So to recap,

1. bp_top / bp_soc system-level
2. FE, BE, ME isolated
3. LCE-CCE compliance testing

muwyse, 2020-03-31:
ME testbenches have been updated / cleaned up and are in regressions as of today (3/31/2020) on dev. LCE-CCE compliance testing is currently enforced via the bp_top testbenches.

dpetrisko, 2020-03-31:
Nice! I'll leave this open until we have FE/BE TBs

dpetrisko, 2020-05-10:
This was mostly sorted out.  Could use a few more testbenches for compliance, but I'm closing the issue

closed 2020-05-10 by dpetrisko