black-parrot/black-parrot #201: Modify spike to trap on instructions BP doesn't support

dpetrisko, 2019-06-09:
(This mod should actually be done in https://github.com/black-parrot/riscv-isa-sim, but I'm putting the issue here because we track bugs here.)

One of the major problems with trace-based diffing with spike is that our partially emulated extensions (e.g. atomics, probably part of muldiv) will not match up with the spike trace. One solution to this is described in #134. Basically consolidating illegal instruction trap into a single committed instruction, but this has the problem of assuming that our emulation code is perfect :)

If we modify spike to throw illegal instructions on our specific unsupported instruction list, then it should match up 1:1. There seems to be an easy path to do this in spike through the instruction list, although my quick experiment had incorrect trap behavior (no mtval) if you simply remove the illegal instructions from the list.

dpetrisko, 2019-09-03:
@farzamgl have you done this for the Linux effort?

farzamgl, 2019-09-03:
> @farzamgl have you done this for the Linux effort?

Yeah [here](https://github.com/black-parrot/riscv-isa-sim/commit/f62e02af93680205ca6146281b09a32265ebca1d)

dpetrisko, 2019-09-03:
Ah, I see.  I'll leave the issue open in case anyone wants to make this more flexible.  I'm thinking maybe a command line parameter --with-emu=[amo_add.d, amo_and.d...]

dpetrisko, 2020-12-22:
We use dromajo now

closed 2020-12-22 by dpetrisko