black-parrot/black-parrot #203: Add MMIO page to start code

dpetrisko, 2019-06-14:
We're moving the test_finish, print and write_to_host_file commands to true MMIO (on the host), so that they'll run on the ASIC.  Because it's MMIO, we need a physical translation for them (1:1 mapping is fine, but not necessary).

I'll update this issue once the host MMIO module is done (but leave in the old print/finish mechanisms).  Then we can add the MMIO page, test that everything works without the legacy mechanism, and then remove them.

dpetrisko, 2019-06-14:
Pushed the module to 'dev_feature_on_chip_network'.  Note that the memory map changed a bit, and now the addresses are per core.  We'll need to modify the newlib code a bit to read mhartid before sending prints or finishes (preferably we'd read mhartid first in some 'init' function and save it, but I'm not familiar enough with the code to know if that's a good idea)

dpetrisko, 2019-06-14:
Further enhancement / possibly easier path: Store the PMA (uncached/cached in TLB).  That way, we can just read from the TLB instead of doing a comparison on every physical translation.  Then we can just add this property to the fake PTW / real PTW instead of manually adding a mapping in software.

farzamgl, 2019-06-17:
Shouldn't the host's base address be in the un-cached range? (high bit = 1)

dpetrisko, 2019-06-17:
That’s no longer the uncached range. The uncached range should be <x8000000 since that’s where all of our devices are

farzamgl, 2019-06-17:
Has it been changed in the mem_top.v?

dpetrisko, 2019-06-17:
It's changed in a local branch that I have, hasn't been merged in yet.  We need to synchronize the software change with the hardware change.  But it shouldn't be too bad since it's just the one-liner that determines cached/uncached

dpetrisko, 2019-06-18:
@farzamgl  Don't worry about this one, I'm lumping it in with a few related changes that I'm doing for MMIO.

closed 2020-03-25 by farzamgl