black-parrot/black-parrot #207: Freeze Verilator gitmodule tracking

muwyse, 2019-06-19:
The Verilator submodule should be frozen at a tagged release and only bumped after we explicitly test the new release. AFAIK, we haven't encountered an issue here, but it seems like it would be better to bump on each release from Verilator instead of tracking master.

This is open for discussion, and the above is simply my opinion.

dpetrisko, 2019-06-19:
I agree we should be more rigorous about testing when we bump. But there are a bunch of bugfixes that happen in master before they propagate to a numbered release.  Especially the versions of Verilator on say Ubuntu apt are very out of date.

But yeah, I could probably set up a manual CI job to build and test all of our external dependencies (gcc, verilator, etc.) to make sure everything works before bumping

dpetrisko, 2021-07-15:
Closing this as we're actively driving verilator development nowadays ;) https://github.com/verilator/verilator/pull/3026

closed 2021-07-15 by dpetrisko