black-parrot/black-parrot #208: Unify tests / split out BP-specific code

dpetrisko, 2019-06-19:
Right now, we've hacked in BP support into beebs, riscv-tests, etc.  This has led to redundant code for things like termination conditions, printing, syscalls, start code, etc.

We would like to have some kind of blackparrot library which has our firmware*, machine-mode syscall implementation, device drivers, magic addresses etc. which we can link from whichever tests we want to build.  This should reduce redundant code and make it easier for users to add new benchmarks.

*Notable exception is riscv-tests, which wants to use it's own firmware for testing. Normally the benchmarks should run in user-mode and then the firmware is a blackbox.  However, these tests must be hacked (or ignored if we find other ways to validate our machine mode behavior (riscv-dv?))

TavioGuarino, 2019-06-19:
I agree. It might be more involved than linking all the test benches to 1 set of support files though. For example, I added some code to syscall.c for beebs to pass ./configure and call the support's main function, and commented out code from syscall.c so it didn't have function aliasing for littlefs_test and for other programs referencing newlib. Also, the linker files have an extra line for the bsg dram end address that is used for code that references newlib but would be undefined for code that doesn't and also the crt.S is different for these programs. The util.h I realized towards the end isn't needed most of the time and encoding.h is always the same.

dpetrisko, 2019-08-14:
#223

closed 2019-08-14 by dpetrisko