black-parrot/black-parrot #217: Synchronous exception priority

dpetrisko, 2019-07-11:
[image]

Synchronous exception priority has changed in the ratified spec.  I _think_ we're actually safe from this, but still need to check.

taylor-bsg, 2019-07-23:
How different is the ratified spec from the most recent drafts of the spec
that we were looking at?

M

dpetrisko, 2019-07-23:
This is the most significant change IMO.  Mostly just clarifications. Other relevant changes:

- Added mcountinhibit which prevents performance counters from ticking (adding now)
- Added new PMP semantics (don't have PMP support yet)
- Removed PLIC description (no PLIC yet)
- Removed requirement that xRET clears LR/SC reservation (already how we implemented)
- Specified extended exception/interrupt code space (we don't have an custom E/I)
- Changed semantics for SUM bit in mstatus.  Originally it let S-mode access all U pages, now it can only access U data pages (We hadn't implemented this yet)

dpetrisko, 2023-07-18:
I believe we are compliant, although I'm sure a user will reopen this when I am proven wrong

closed 2023-07-18 by dpetrisko