black-parrot/black-parrot #223: Port all tests to use libperch

dpetrisko, 2019-07-16:
Currently, all of our tests have hardcoded host interfaces (printing, termination, firmware generally). I've already ported some of our demos and it seems pretty painless
 
This way we can just like a static library and get all of our host interface for free (so that when it changes slightly all of our tests don't break).

dpetrisko, 2019-07-16:
Note: Nominally the stack is Application (user-level) -> PanicRoom (user-level)-> libperch (firmware).  So users most likely won't be making direct calls into the firmware.  An example of direct use would be the kernel querying hartid.

taylor-bsg, 2019-07-23:
What is libperch?

dpetrisko, 2019-07-23:
Firmware library for emulation, mainly, but generally for anything we want to include in M-mode

taylor-bsg, 2019-07-23:
Cool. What does perch mean? Like the fish?

dpetrisko, 2019-07-23:
No, like a perch that a BlackParrot roosts on

taylor-bsg, 2019-07-23:
Ah got it!

dpetrisko, 2019-08-14:
> I agree. It might be more involved than linking all the test benches to 1 set of support files though. I added some code to syscall.c for beebs to pass ./configure and call the support's main function. I also commented out code from syscall.c so it didn't have function aliasing for littlefs_test and for other programs referencing newlib. The linker files have an extra line for the bsg dram end address that is used for code that references newlib but would be undefined for code that doesn't. The crt.S is also slightly different for these programs. However, the util.h isn't needed most of the time and encoding.h is always the same.

_Originally posted by @TavioGuarino in https://github.com/black-parrot/pre-alpha-release/issues/208#issuecomment-503770435_

We need to modify the linker script in libperch to be more general...

closed 2026-04-03 by dpetrisko