black-parrot/black-parrot #225: Combine LCE-CCE networks

dpetrisko, 2019-07-18:
Currently, we have 5 unidirectional networks (CCE-LCE).  However, since every CCE can communicate with every LCE, the links on the mesh need to be bi-directional.

Proposal: Consolidate to 3 networks: req, cmd(cmd+data_cmd), resp(resp+data_resp).  I think deadlock proofs are easy enough to reason about since there should be no more conflicting traffic on these combined networks (every LCE only has _either_ an outstanding cmd/data_cmd or resp/data_resp)

taylor-bsg, 2019-07-18:
I agree that we should examine consolidation. I think it is useful to look
at what the margin performance benefit we get from having 5 separate
networks, and also from being wide on each link. As I think Mark and Dan
have clearly identified, the issue may not be so much the overhead of the
LCE connection, but rather than the inter-router links. The other thing I
have noticed is that our gearboxs that go between different networks by
expanding out to the full 512 bits take a lot of area and increase latency
and it would more preferable use an incremental approach where we can do a
word at a time. This applies also to reading data out of the caches.

The other reason we want to reduce wires is so that we can move to a mesh,
which will be more physicaly design friendly and reduce latency, but will
require more wires per tile.

dpetrisko, 2019-07-18:
> The other thing I
> have noticed is that our gearboxs that go between different networks by
> expanding out to the full 512 bits take a lot of area and increase latency
> and it would more preferable use an incremental approach where we can do a
> word at a time.

Yep, Mark and I had a conversation about that recently, too.  Basically, the current thought is to do it _after_ we've consolidated, since we will need to write a module to read the headers of the messages to decide what to do with them.  Another initial constraint that we've considered is to have the link size of wormhole >= header_flit_size.

As far as the wide link benefits, I think it's interesting.  Once the wh networks are set up properly, we could easily sweep (up to num flits = 1) over link size to see performance / routing tradeoffs.  I think non-wh link studies have limited usefulness given their PD impact.

dpetrisko, 2019-08-14:
Done in dev

closed 2019-08-14 by dpetrisko