black-parrot/black-parrot #227: Documentation required

BaudouinChauviere, 2019-07-18:
I am trying to use your rtl at the integration exercise in order to feed it to our project OpenFPGA.

Even with the help of one of your students I am having issues getting the Verilog needed for our project.

It seems like you had a larger documentation before the pre-alpha release and that now the README on the front page actually gives very little information compared to what you had beforehand. Could you update that please? 

You can contact me for faster collaboration

dpetrisko, 2019-07-18:
Apologies for the lack of documentation.

What are you trying to do precisely?

dpetrisko, 2019-07-18:
```
cd pre-alpha-release
make tools -j
make progs ucode
```
will initialize the repo (this may take a while depending on how many cores you have, it builds a gcc toolchain along the way).

```
cd bp_top/syn
make build.v sim.v
```

will run a simple riscv unit test

```
cd bp_top/syn
make build.v sim.v CFG=e_bp_oct_core_cfg PROG=hello_world
```

will run an 8-core hello_world program

BaudouinChauviere, 2019-07-18:
I tried doing what you advised me to do and here is the result:
[image]
[image]
If you want I can come to talk to you to fix this together

dpetrisko, 2019-07-18:
```fatal: could not get a repository handle for submodule 'riscv-glibc'```

This is the first error, the rest are results of not cloning the toolchain.  Possibly you have changed some state in the repo trying to get this to work.  The toolchain has been extensively tested on CentOS 7. What OS are you running?

Also try this to get the repo to a clean state:
```
cd pre-alpha-release
git submodule deinit -f .
git submodule update --init --recursive
```

dpetrisko, 2019-07-18:
I also ask about what you are doing, because the toolchain builds are only necessary to actually build and run simulations in Verilator or VCS.  If you are simply trying to map the design to your FPGA, I suggest using https://github.com/bespoke-silicon-group/bsg_sv2v to pickle the design to a single pure Verilog file and run that through.  Else, check out the bp_top/syn/flist.vcs for the full file list

taylor-bsg, 2019-07-23:
Just following up on this, Baudouin -- were you able to resolve the issue?

M

BaudouinChauviere, 2019-07-23:
Hi Michael,

we resolved the issue but the issue is that congestion is too big for our program and it dies before implementing it. 
My two cents on that issue though: I think your application is memory hungry and I am currently implementing it with an homogeneous fabric (no time to make the suitable architecture). With that said I think if we customize the fabric to your need, we can make it work. As for now, we are doing the verification of new hard logic (memory now, other blocks will come soon) so I don't think we can make it work under a short notice. 

Here is the 
[Debug_BP.txt](https://github.com/black-parrot/pre-alpha-release/files/3424366/Debug_BP.txt)

Thanks for the follow up

BaudouinChauviere, 2019-07-23:
As a side note, it is actually before the extension we have since it dies during VTR's routing

dpetrisko, 2019-07-23:
Yeah, the major memories in the system are 32kB I$, 32kB D$, 4kB directory tags, 2kB register file.  So it would be pretty heavy without hardened RAMs.  It's theoretically possible to shrink the caches to 8K, which would also reduce the directory tag size.  However, this is untested and would require some bug-hunting to get working.

jonmasters, 2019-08-13:
Circling back to the original topic of this thread, on documentation. There's almost none today. Figuring out the project requires one to start at bp_top and work one's way down. Could this be fixed?

dpetrisko, 2019-08-13:
Thanks for the comment, Jon.

Our 1.0 release will certainly have documentation, both for microarchitecture specifications and "out-of-the-box" instructions for simulation.  Our pre-alpha documentation had grown fairly stale on both of these fronts as the project evolved, so it's a matter of updating it to reflect reality. The good news is that BlackParrot has become much more stable -- we should be able to maintain the next release of documentation moving forward.

taylor-bsg, 2019-08-14:
Thanks for the feedback, Jon — very helpful.

We would love to learn more about your interest in BP.

In terms of the V 1.0, we are currently getting Linux running on BP and
will release 1.0 (with updates docs) when that is stable. We also recently
taped out BP in 12nm GF technology.

M

dpetrisko, 2019-10-02:
Thanks for your patience on this issue.

Following up, documentation is in progress (#266).  We hope to release v1.0 soon, which will correspond with the new documentation.

dpetrisko, 2020-04-26:
Closing this issue as v1 has much more documentation.  @jonmasters @BaudouinChauviere let us know if there are any specific areas that could use more elucidation

closed 2020-04-26 by dpetrisko