black-parrot/black-parrot #228: Integrate black-parrot into LiteX and LiteX-BuildEnv

mithro, 2019-07-18:
[LiteX](https://github.com/enjoy-digital/litex) is an FPGA SoC generator that supports a wide range of FPGA boards and platforms. It currently support multiple soft-cpus, including multiple RISC-V implementations. You can find the details about supported CPUs on this wiki page -> https://github.com/enjoy-digital/litex/wiki/Soft-CPU

It would be awesome to see Black-Parrot in the LiteX system. It would let us very easily compare the resource usage of Black-Parrot to Rocket and VexRISCV.

[LiteX-BuildEnv](https://github.com/timvideos/litex-buildenv) is my environment which uses LiteX to create images for many different FPGA boards. It supports Zephyr, Linux, Micropython and bare metal firmware.

You can find a diagram of this below;
[image]

There is also a [Linux on LiteX VexRISCV](https://github.com/enjoy-digital/linux-on-litex-vexriscv) repository which would be cool to port to use Black-Parrot as an alternative option.

You can find the existing CPUs at https://github.com/enjoy-digital/litex/tree/master/litex/soc/cores/cpu
Rocket is probably the closest configuration which can be found at https://github.com/enjoy-digital/litex/blob/master/litex/soc/cores/cpu/rocket/core.py

taylor-bsg, 2019-07-18:
Nice we will take a look!

mithro, 2019-07-28:
@taylor-bsg If you need any help figuring out how to do this, feel free to reach out.

mithro, 2020-01-14:
FYI - This work in in progress at https://github.com/enjoy-digital/litex/pull/278

taylor-bsg, 2021-12-16:
Completed.

closed 2021-12-16 by taylor-bsg