black-parrot/black-parrot #234: Prevent grabbing stale CCE ucode

dpetrisko, 2019-08-02:
Currently, we generate all of the CCE ucode binaries as part of the repo setup.  This leads to annoyances like when you change branches, you often have to go through the setup again.  And if you forget, the code just won't work.

I see two options:
1) Generate the ucode during the compilation flow based on the parameters needed. This is a super quick step, so it's not much overhead.  It takes responsibility away from users, which I like
2) Tag the ucode with some sort of version information, which is checked during the flow, so we can error out appropriately.  

2 may be a good idea anyway.  Basically a magic number at the header which the CCE checks (new instruction) to ensure valid firmware is being loaded.  This requires a bit of hardware overhead, but is a feature that real-world processors probably have.

taylor-bsg, 2019-08-02:
I think having the CCE check is putting the complexity in hardware. Maybe
it is better just to have a CCE ISA version # (or a git tag) that is
compiled in as a memory mapped address on the CLINT. Then the boot code can
read the CCE ISA version number (and any other processor version) from the
outside and compare to the corresponding number of the ucode that it is
trying to load?

dpetrisko, 2019-08-02:
Good idea! We can certainly add it to the boot code.  

I believe there's some security advantage of a hardware check: you could prevent users from uploading unsigned ucode.  Whereas with this solution, users could flash their own firmware and get around it.  Not saying that's a use case we should prioritize, but something to consider

dpetrisko, 2023-07-18:
Closing as stale; could be a useful feature in industry...

closed 2023-07-18 by dpetrisko